carla-simulator/carla #6988: UE4.27 installed, but UE5.3 launches

guptaPrabhav, 2023-12-08:
Hey everyone.

This question is more oriented towards the Unreal Engine. But the only reason I was downloading and installing Unreal Engine was for running CARLA. Another reason as to why I am asking this question here is because the github repo for UE has it's issues page dissabled.

Platform/OS: Ubuntu 20.04 LTS
Problem you have experienced: Installed UE4.27, but on launching, UE5.3 launches
What you expected to happen: UE4.27 to launch
Steps to reproduce: I followed [this](https://docs.unrealengine.com/4.27/en-US/SharingAndReleasing/Linux/BeginnerLinuxDeveloper/SettingUpAnUnrealWorkflow/) document (as is also mentioned in the CARLA docs) to install and setup UE.
Other information (documentation you consulted, workarounds you tried): 

Things to note here:
During the step where we are supposed to clone the repository, I specified the version for UE, which was 4.27, in the command itself. Once the cloning was complete, I proceeded with the remaining steps. Once the build was complete, the UnrealEditor that launches, displayed version 5.3.2 for UE.

Is there a way I can chage the UE version installed now? If I have to re-install, is there some documentation that I can follow that spicifes how to download and install version 4.27.

I need to install UE 4.27 on my system in order to run CARLA. Any and all help will be greatly appreciated.

Thanks!

PatrickPromitzer, 2023-12-11:
The current version of Carla is running in a forked UE 4.26 version.
The normal Unreal Engine Version doesn't work.
```commandline
 git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git ~/UnrealEngine_4.26
```

Carla is using a variable to get the right Unreal Engine version
```commandline
 export UE4_ROOT=~/UnrealEngine_4.26
```
The documentation tells you to add it in your .bashrc file, not to have to set the variable every time.

Carla documentation: https://carla.readthedocs.io/en/latest/build_linux/

guptaPrabhav, 2023-12-12:
Thanks @PatrickPromitzer !

I followed the above steps and it Carla_UE4 is installed now, and runs fine also. 

Thanks once again for the response!

closed 2023-12-12 by guptaPrabhav