carla-simulator/carla #6989: fatal error C1083: Cannot open include file: 'carla/Version.h': No such file or directory [C:\carla\Build\libcarla-visualstudio\LibCarla\cmake\client\carla_client_debug.vcxproj]

nadrainam, 2023-12-11:
Hi
I am trying to install carla 0.9.11 version with the visual studio 2017
I am facing the following errors. (please see the attached file).
Can someone please help me to resolve this issue.
Best Regards
Nadra Inam
Germany
[carla error.txt](https://github.com/carla-simulator/carla/files/13634894/carla.error.txt)

-->

javageek68, 2024-02-08:
@nadrainam  Did you ever find a fix for this?  I'm running into the same issue.

haorank, 2024-04-09:
@nadrainam Hi, I've encountered the same issue. Have you managed to solve it?

PC12138XD, 2024-04-13:
I got the same problem. Did you ever find a fix for this?

haorank, 2024-04-14:
The following link might be helpful to you：
https://link.zhihu.com/?target=https%3A//github.com/carla-simulator/carla/issues/6248