carla-simulator/carla #7004: Manual_control.py crashing with rpc server issue

dev3225, 2023-12-14:
The script starts ->  then opens pygame window -> blank screen -> not responding message on titlebar - > finally crashes and throws runtime error as shown below

C:\carla\PythonAPI\examples>python manual_control.py
pygame 2.4.0 (SDL 2.26.4, Python 3.9.0)
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO: listening to server 127.0.0.1:2000

Welcome to CARLA manual control.

Use ARROWS or WASD keys for control.

    W            : throttle
    S            : brake
    A/D          : steer left/right
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot
    M            : toggle manual transmission
    ,/.          : gear up/down
    CTRL + W     : toggle constant velocity mode at 60 km/h

    L            : toggle next light type
    SHIFT + L    : toggle high beam
    Z/X          : toggle right/left blinker
    I            : toggle interior light

    TAB          : change sensor position
    ` or N       : next sensor
    [1-9]        : change to sensor [1-9]
    G            : toggle radar visualization
    C            : change weather (Shift+C reverse)
    Backspace    : change vehicle

    O            : open/close all doors of vehicle
    T            : toggle vehicle's telemetry

    V            : Select next map layer (Shift+V reverse)
    B            : Load current selected map layer (Shift+B to unload)

    R            : toggle recording images to disk

    CTRL + R     : toggle recording of simulation (replacing any previous)
    CTRL + P     : start replaying last recorded simulation
    CTRL + +     : increments the start time of the replay by 1 second (+SHIFT = 10 seconds)
    CTRL + -     : decrements the start time of the replay by 1 second (+SHIFT = 10 seconds)

    F1           : toggle HUD
    H/?          : toggle help
    ESC          : quit

INFO:  Found the required file in cache!  Carla/Maps/Nav/Town10HD_Opt.bin
Traceback (most recent call last):
  File "C:\carla\PythonAPI\examples\manual_control.py", line 1383, in <module>
    main()
  File "C:\carla\PythonAPI\examples\manual_control.py", line 1375, in main
    game_loop(args)
  File "C:\carla\PythonAPI\examples\manual_control.py", line 1273, in game_loop
    controller = KeyboardControl(world, args.autopilot)
  File "C:\carla\PythonAPI\examples\manual_control.py", line 385, in __init__
    world.player.set_autopilot(self._autopilot_enabled)
RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error.

PatrickPromitzer, 2023-12-15:
I can be wrong, but a bind error could happen if the port is used from a different software.
https://carla.readthedocs.io/en/latest/python_api/#commandsetautopilot

The documentation says that the default port is 8000
I would look if that port is free.

dev3225, 2024-01-10:
I checked the port usage and 8000 is free but it till throws the same error. 

> I can be wrong, but a bind error could happen if the port is used from a different software. https://carla.readthedocs.io/en/latest/python_api/#commandsetautopilot
> 
> The documentation says that the default port is 8000 I would look if that port is free.

dev3225, 2024-01-25:
It worked without issues when I commented out autopilot module in the python script.

closed 2024-01-25 by dev3225