carla-simulator/carla #7008: Cannot build the Unreal Engine 4.26 Carla branch from source files

Zhenlin-Xu, 2023-12-15:
CARLA version: 0.9.15
Platform/OS: Windows 10
Problem you have experienced: Cannot build the Unreal Engine 4.26 from source files.
What you expected to happen: I hope to compile the UE4 from source files.
Steps to reproduce: I just run the two .bat scripts and build the UE4 project in Visual Studio 2019 as the documentation instructs.
Other information (documentation you consulted, workarounds you tried):

Hello! I am trying to compile the UE4.26 from scratch and then compile CARLA. However, there are some errors when I try to compile it.

I am using the official repository provided in the CARLA documentation:
```
git clone --depth 1 -b carla https://github.com/CarlaUnreal/UnrealEngine.git .
```

Here is the error I encounter:
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	C2487	'SpillCellID': member of dll interface class may not be declared with dll interface	UE4	
U:\UnrealEngine\Engine\Plugins\Experimental\GeometryProcessing\Source\GeometricObjects\Public\Spatial\SparseDynamicOctree3.h	291	

Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB3073	The command "..\..\Build\BatchFiles\Build.bat -Target="UE4Editor Win64 Development" -Target="ShaderCompileWorker Win64 Development -Quiet" -WaitMutex -FromMsBuild" exited with code 6.	UE4	
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.MakeFile.Targets	45	
```

[image]

ronyshaji, 2023-12-17:
@Zhenlin-Xu I am facing the same issue. Did you get any resolution?

Zhenlin-Xu, 2023-12-17:
Nop.

ronyshaji, 2023-12-17:
@Zhenlin-Xu I think the Unreal Engine source code has been changed from the repo, and it no longer builds with the new branch. Maybe it's a bug. Can you add the bug tag to this post so that somebody checks it?

Zhenlin-Xu, 2023-12-17:
I think only the contributor of the repo can add a tag/label to the issue. Am I correct or is there another way to add the label by myself.

ronyshaji, 2023-12-17:
@germanros1987 Can you please have a look into this issue?

Patrone411, 2023-12-17:
haveing the exact same issue.

andrejfsantos4, 2023-12-18:
I have the same issue.

A temporary workaround which seems to work for me is to revert one of the changes in [this commit](https://github.com/CarlaUnreal/UnrealEngine/commit/c48d13957dd607bd6f42dde61140c70ec4eca24e):
Change the line 138 of _Engine/Plugins/Experimental/GeometryProcessing/Source/GeometricObjects/Public/Spatial/SparseDynamicOctree3.h_ from
`class GEOMETRICOBJECTS_API FSparseDynamicOctree3`
back to
`class FSparseDynamicOctree3`

However I face other issues when trying to build the CARLA server, as mentioned in #7012.

Blyron, 2023-12-19:
We are aware of this we should be fixing soon

chirudeep001, 2023-12-19:
@andrejfsantos4  The issue was resolved after 
Change the line 138 of Engine/Plugins/Experimental/GeometryProcessing/Source/GeometricObjects/Public/Spatial/SparseDynamicOctree3.h from
class GEOMETRICOBJECTS_API FSparseDynamicOctree3
back to
class FSparseDynamicOctree3

but i am encountering new error when i do make pythonAPI

andrejfsantos4, 2023-12-19:
> @andrejfsantos4 @Blyron The issue was resolved after Change the line 138 of Engine/Plugins/Experimental/GeometryProcessing/Source/GeometricObjects/Public/Spatial/SparseDynamicOctree3.h from class GEOMETRICOBJECTS_API FSparseDynamicOctree3 back to class FSparseDynamicOctree3
> 
> but i am encountering new error when i do make pythonAPI

@chirudeep001 Sorry, for my application I just installed the client with `pip install carla` since I don't need to make custom changes to it, so I can't help you there

Blyron, 2023-12-19:
Should be fixed now with the merge of 
https://github.com/CarlaUnreal/UnrealEngine/pull/16 into Engine code

ronyshaji, 2023-12-20:
@Zhenlin-Xu It's building now. You can close the issue.

closed 2023-12-20 by Zhenlin-Xu

Bniultu, 2024-03-17:
Hi ,all .
when i installation Carla editor there has some error can any one help me with that.
(>C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.MakeFile.Targets(45,5): error MSB3073: The command "..\..\Build\BatchFiles\Build.bat -Target="UE4Editor Win64 Development" -Target="ShaderCompileWorker Win64 Development -Quiet" -WaitMutex -FromMsBuild" exited with code 6.
1>Done building project "UE4.vcxproj" -- FAILED.
========== Build: 0 succeeded, 1 failed, 2 up-to-date, 0 skipped ==========)