carla-simulator/carla #7011: Digital twin tool crash

zjr-bit, 2023-12-18:
CARLA version: 0.9.15
Platform/OS: ubuntu18.04 LTS
Problem you have experienced: I followed the official instructions, and when I input the processed .osm file in the interface, and meet the crash. The error infos are as follows:

```
Fatal error: [File:/home/jiawei/carla/Unreal/CarlaUE4/Plugins/CarlaTools/Source/CarlaTools/Private/CarlaTools.cpp] [Line: 33] Exception thrown: connect: Connection refused [system:111 at /home/jiawei/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/detail/reactive_socket_service.hpp:587:33 in function 'connect'] << callstack too long >>

libUE4Editor-Core.so!FGenericPlatformMisc::RaiseException(unsigned int) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/GenericPlatform/GenericPlatformMisc.cpp:472]

libUE4Editor-Core.so!FOutputDevice::LogfImpl(char16_t const*, ...) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Core/Private/Misc/OutputDevice.cpp:61]

libUE4Editor-CarlaTools.so!carla::throw_exception(std::exception const&) [/home/jiawei/carla/Unreal/CarlaUE4/Plugins/CarlaTools/Source/CarlaTools/Private/CarlaTools.cpp:33]

libUE4Editor-CarlaTools.so!boost::throw_exception(std::exception const&)(+0x5)

libUE4Editor-CarlaTools.so!boost::throw_exception(std::exception const&, boost::source_location const&)(+0x5)

libUE4Editor-CarlaTools.so!boost::asio::basic_socket<boost::asio::ip::tcp, boost::asio::any_io_executor>::connect(boost::asio::ip::basic_endpoint<boost::asio::ip::tcp> const&) [/home/jiawei/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaDependencies/include/boost/asio/basic_socket.hpp:864]

libUE4Editor-CarlaTools.so!UMapPreviewUserWidget::ConnectToSocket(FString, FString, int) [/home/jiawei/carla/Unreal/CarlaUE4/Plugins/CarlaTools/Source/CarlaTools/Private/MapPreviewUserWidget.cpp:55]

libUE4Editor-CarlaTools.so!UMapPreviewUserWidget::execConnectToSocket(UObject*, FFrame&, void*) [/home/jiawei/carla/Unreal/CarlaUE4/Plugins/CarlaTools/Intermediate/Build/Linux/B4D820EA/UE4Editor/Inc/CarlaTools/MapPreviewUserWidget.gen.cpp:74]

libUE4Editor-CoreUObject.so!UFunction::Invoke(UObject*, FFrame&, void*) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/Class.cpp:5588]

libUE4Editor-CoreUObject.so!UObject::CallFunction(FFrame&, void*, UFunction*) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:997]

libUE4Editor-CoreUObject.so!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1094]

libUE4Editor-CoreUObject.so!void ProcessScriptFunction<void (*)(UObject*, FFrame&, void*)>(UObject*, UFunction*, FFrame&, void*, void (*)(UObject*, FFrame&, void*)) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:895]

libUE4Editor-CoreUObject.so!ProcessLocalFunction(UObject*, UFunction*, FFrame&, void*) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1128]

libUE4Editor-CoreUObject.so!ProcessLocalScriptFunction(UObject*, FFrame&, void*) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1094]

libUE4Editor-CoreUObject.so!UFunction::Invoke(UObject*, FFrame&, void*) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/Class.cpp:5588]

libUE4Editor-CoreUObject.so!UObject::ProcessEvent(UFunction*, void*) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/CoreUObject/Private/UObject/ScriptCore.cpp:1990]

libUE4Editor-UMG.so!void TScriptDelegate<FWeakObjectPtr>::ProcessDelegate<UObject>(void*) const [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/UObject/ScriptDelegates.h:247]

libUE4Editor-UMG.so!void TMulticastScriptDelegate<FWeakObjectPtr>::ProcessMulticastDelegate<UObject>(void*) const [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/UObject/ScriptDelegates.h:493]

libUE4Editor-UMG.so!UButton::SlateHandleClicked() [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/UMG/Private/Components/Button.cpp:203]

libUE4Editor-UMG.so!TBaseUObjectMethodDelegateInstance<false, UButton, FReply (), FDefaultDelegateUserPolicy>::Execute() const [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:593]

libUE4Editor-Slate.so!SButton::ExecuteOnClick() [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Slate/Private/Widgets/Input/SButton.cpp:385]

libUE4Editor-Slate.so!SButton::OnMouseButtonUp(FGeometry const&, FPointerEvent const&) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Slate/Private/Widgets/Input/SButton.cpp:304]

libUE4Editor-Slate.so!FSlateApplication::RoutePointerUpEvent(FWidgetPath const&, FPointerEvent const&) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Slate/Private/Framework/Application/SlateApplication.cpp:4816]

libUE4Editor-Slate.so!FSlateApplication::ProcessMouseButtonUpEvent(FPointerEvent const&) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Slate/Private/Framework/Application/SlateApplication.cpp:5356]

libUE4Editor-Slate.so!FSlateApplication::OnMouseUp(EMouseButtons::Type, FVector2D) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Slate/Private/Framework/Application/SlateApplication.cpp:5321]

libUE4Editor-ApplicationCore.so!FLinuxApplication::ProcessDeferredMessage(SDL_Event) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/ApplicationCore/Private/Linux/LinuxApplication.cpp:370]

libUE4Editor-ApplicationCore.so!FLinuxPlatformApplicationMisc::PumpMessages(bool) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/ApplicationCore/Private/Linux/LinuxPlatformApplicationMisc.cpp:428]

UE4Editor!FEngineLoop::Tick() [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4748]

UE4Editor!GuardedMain(char16_t const*) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/Launch.cpp:171]

libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/home/jiawei/UnrealEngine_4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]

libc.so.6!__libc_start_main(+0xe6)

UE4Editor!_start()
```

What you expected to happen: Generate the map without error
Steps to reproduce: follow the [official instructions](https://carla.readthedocs.io/en/0.9.15/build_update/#launch-the-server). And input the processed map file `carla/Build/libosmscout-source/maps/map-09` , then click 'config' and get the error.
Could anyone give me some tips? Thanks very much!

nusstu0019, 2024-01-18:
hello, when I put the osm file into maps file and run ./build.sh, it says that not found and Cannot find Import executable! Do you meet the same problem?

Blyron, 2024-05-31:
Look at this issue please and discuss there

closed 2024-05-31 by Blyron

chaoer, 2024-06-27:
same problem.

have you figured out what happened?

berberlol, 2024-07-04:
same error in my pc,is there any ways to solve? @zjr-bit   @Blyron