carla-simulator/carla #7012: CARLA 0.9.15 fails to build Houdini Engine

andrejfsantos4, 2023-12-18:
**CARLA version:** 0.9.15
**Platform/OS:** Windows 10 - Visual Studio 2022

**Problem you have experienced:** 
After running Update.bat, I get this error
```
-[BuildCarlaUE4]: Building Unreal Editor...
Using 'git status' to determine working set for adaptive non-unity build (D:\Programs\UnrealEngine-0.9.15).
Creating makefile for CarlaUE4Editor (UnrealBuildTool.exe is newer)
While compiling D:\Programs\carla_0.9.15\Unreal\CarlaUE4\Intermediate\Build\BuildRules\CarlaUE4ModuleRules.dll:
d:\Programs\carla_0.9.15\Unreal\CarlaUE4\Plugins\HoudiniEngine\Source\HoudiniEngine\HoudiniEngine.Build.cs(44,7) : error CS0246: The type or namespace name 'EpicGames' could not be found (are you missing a using directive or an assembly reference?)
UnrealBuildTool: ERROR: Unable to compile source files.
```
which I attempted to fix by replacing `using EpicGames.core` with `using Tools.DotNETCommon` as suggested [here](https://dev.epicgames.com/community/snippets/7YD/ue4-to-ue5-plugin-upgrade-notes) (not sure if this is the correct fix). After this, I get a different error:

```
-[BuildCarlaUE4]: [Batch params]: --build --all
--build
--all
ECHO is off.
-[BuildCarlaUE4]: Building Unreal Editor...
Using 'git status' to determine working set for adaptive non-unity build (D:\Programs\UnrealEngine-0.9.15).
Creating makefile for CarlaUE4Editor (UnrealBuildTool.exe is newer)
Houdini Engine : Houdini 20.0.556 could not be found. Houdini Engine will not be available in this build.
UnrealBuildTool: ERROR: Could not find definition for module 'EditorFramework', (referenced via Target -> CarlaTools.Build.cs -> HoudiniEngine.Build.cs -> HoudiniEngineRuntime.Build.cs)
make: *** [CarlaUE4Editor] Error 6
```

andrejfsantos4, 2023-12-20:
I should add that if I remove Houdini Engine and all references to it, the build finishes successfully and I'm able to run CARLA 0.9.15

PierceG6, 2024-02-03:
> I should add that if I remove Houdini Engine and all references to it, the build finishes successfully and I'm able to run CARLA 0.9.15

hihi can you please explain how you did it? I have the same issue.

andrejfsantos4, 2024-02-06:
> > I should add that if I remove Houdini Engine and all references to it, the build finishes successfully and I'm able to run CARLA 0.9.15
> 
> hihi can you please explain how you did it? I have the same issue.

Hi @PierceG6,
This was a while ago but if I remember correctly these were the changes - I removed the files in red, and made these changes to the files with the yellow icon. Then rebuild and hopefully it'll work!

[image]

[image]

2002atulyadav, 2024-03-06:
I have builded the carla && unreal engine successfully but when i am running the "make Launch" i facing bellow issue:
Unreal engine 4.26
carla 0.9.15
ubuntu 20.04 with all required prerequsites

[image]
[image]
[image]

xavisolesoft, 2024-04-19:
We deprecated houdini en CARLA, if you get the last dev code would not have issues with houdini anymore.

closed 2024-04-19 by xavisolesoft

asina13, 2024-06-21:
> I have builded the carla && unreal engine successfully but when i am running the "make Launch" i facing bellow issue: Unreal engine 4.26 carla 0.9.15 ubuntu 20.04 with all required prerequsites
> 
> ![Screenshot from 2024-03-06 12-00-50](https://private-user-images.githubusercontent.com/122152736/310437609-20411967-f735-4ab4-9d63-ef01f9394256.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg5NzA2NTEsIm5iZiI6MTcxODk3MDM1MSwicGF0aCI6Ii8xMjIxNTI3MzYvMzEwNDM3NjA5LTIwNDExOTY3LWY3MzUtNGFiNC05ZDYzLWVmMDFmOTM5NDI1Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyMVQxMTQ1NTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNGNmZDliZmU1YzFhM2YxYzU3NWEyNjIyNGIxZWJhOTZjYTcwOTNjNWQ3NDE1YjI4YzljNjViMDliYTI2YTg3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.buh_KwvAUGo7cfRL4xPfzGnMx8uBK0-_m1B3n5NFdjY) ![Screenshot from 2024-03-06 14-48-58](https://private-user-images.githubusercontent.com/122152736/310437616-cdfebb99-effb-44a9-b704-bcb9478b04aa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg5NzA2NTEsIm5iZiI6MTcxODk3MDM1MSwicGF0aCI6Ii8xMjIxNTI3MzYvMzEwNDM3NjE2LWNkZmViYjk5LWVmZmItNDRhOS1iNzA0LWJjYjk0NzhiMDRhYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyMVQxMTQ1NTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hODg5YmE2NDA3YWViMTY4NGNjNjZmODQ1Nzc0YjZjNjM1NGYzMzIxMDRkYTg2OTVjZTZmN2ViZGMyNGE0MGM4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.UmAh2ttzvK2zRvFphkLBXX4s0c-uwnJeMgnXbS3ULSc) ![Screenshot from 2024-03-06 14-49-35](https://private-user-images.githubusercontent.com/122152736/310437626-346965c6-e40f-46df-a2ba-afa600979806.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg5NzA2NTEsIm5iZiI6MTcxODk3MDM1MSwicGF0aCI6Ii8xMjIxNTI3MzYvMzEwNDM3NjI2LTM0Njk2NWM2LWU0MGYtNDZkZi1hMmJhLWFmYTYwMDk3OTgwNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYyMVQxMTQ1NTFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MDVmNmI3ZTNlMmM4Y2YzMjU3ZDJlNzUxMzNiZTQ1ZjVlODM4YmRmNDk5MDVkZDYyODI2OWE1MjkzNzA1ZTQ0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.6US2re8zXnS8bD41500KjJX51FuC0B9LF7xZzcjateY)

did you solve the problem? ım facing the same issue