carla-simulator/carla #7013: Digital Twin Tool - the documentation doesn't add up with the files

PatrickPromitzer, 2023-12-18:
CARLA version: 0.9.15 (git version)
Platform/OS: Ubuntu 22.04

The documentation doesn't add up with the existing files.
https://carla.readthedocs.io/en/0.9.15/adv_digital_twin/

```
Downloading an OSM file and navigating in the interface

...
cd {CARLA_ROOT}/Build/libosmcout-source/maps/
./build.sh <path_to_osm_file>
```
[image]

The file "Import/src/Import" doesn't exist in the "libosmscout-source" folder, but it exists in the "libosmscout-build" folder.

```
OpenStreetMap browser
To open the OSM browser, open the content browser and navigate to CarlaToolsContent/OnroadMapGenerator.
Right click on UW_OnRoadMainWidget and select Launch Editor Utility Widget from the context menu. 
This will open the tool's interface.
```
The "Unreal/CarlaUE4/Content/CarlaToolsContent/" doesn't exist, and the only place the folder is mentioned in the documentation is the "Digital Twin Tool" page.

InigoAgirre, 2024-01-03:
Hello @PatrickPromitzer, I have the same problem, did you manage to solve it?

PatrickPromitzer, 2024-01-08:
@InigoAgirre I didn't look in the missing Content folders but fixed the tool error.

You can make a soft link for "Import/src/Import" from "libosmscout-build" to "libosmscout-source" 

Ubuntu
ln -s TARGET LINK_NAME
```commandline
mkdir {CARLA_ROOT}/Build/libosmscout-source/Import
mkdir {CARLA_ROOT}/Build/libosmscout-source/Import/src
ln -s {CARLA_ROOT}/Build/libosmscout-build/Import/Import-1.1.1 {CARLA_ROOT}/Build/libosmscout-source/Import/src/Import
```
I can't check if the path is right, because I need to build the tool again (and I got an error building it).

matejm42, 2024-01-08:
I was able to use the Digital Twin Tool without the libosmcout - see #7053

But I was not able to use `libosmcout-source/maps/build.sh

> /mnt/ssd/matejm42/carla_0.9.15/Build/libosmscout-source/maps$ ./build.sh map.osm 
./build.sh: 8: [[: not found
./build.sh: 10: [[: not found
./build.sh: 12: [[: not found
Cannot find Import executable!

Blyron, 2024-05-31:
> CARLA version: 0.9.15 (git version) Platform/OS: Ubuntu 22.04
>
> The documentation doesn't add up with the existing files. https://carla.readthedocs.io/en/0.9.15/adv_digital_twin/
>
> ```
> Downloading an [...]

To be able to see the CarlaTools Content you have to click on "view options" (bottom right of the content browser) and enable see plugin content.

Blyron, 2024-05-31:
> I was able to get the utlity to work by changing the top of the file from `#!/bin/sh` to `#!/bin/bash`
> 
> From doing some reading, [[ is not supported by shell.
> 
> Additionally I had to create a symbolic link between the build and the src folders `ln -s {CARLA_ROOT}/Build/libosmscout-build/Import/Import-1.1.1 {CARLA_ROOT}/Build/libosmscout-source/Import/src/Import`
> 
> But I cannot get any further then that without UE4Editor crashing when trying to open the output.

About this we would need callstack and know if you followed the proper instructions by, starting the renderer, the click on config and then render.

berberlol, 2024-07-04:
ubuntu20.04 carla0.9.15 same issue with above comments

M-Colley, 2026-10-26:
Draft PR #10019 corrects the `ue4-dev` Digital Twin documentation's renderer paths, Bash command, and plugin-visibility instructions. It currently has merge conflicts. The tool has not been validated in a live Unreal editor.