carla-simulator/carla #7017: dist.carla.org - connection timed out

nenadilic84, 2023-12-20:
CARLA version: 0.9.15
Platform/OS: Ubuntu 20.04
Problem you have experienced: Not able to connect to [dist.carla.org:80](http://dist.carla.org/) (34.227.255.250), connection timed out 
What you expected to happen: successful connection 
Steps to reproduce: 
```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 1AF1527DE64CB8D9
sudo add-apt-repository "deb [arch=amd64] http://dist.carla.org/carla $(lsb_release -sc) main"
```
Output:
```
Err:10 http://dist.carla.org/carla focal InRelease                                                                                                                                                                          
  Could not connect to [dist.carla.org:80](http://dist.carla.org/) (34.227.255.250), connection timed out
Fetched 115 kB in 30s (3.788 B/s)                
Reading package lists... Done
W: Failed to fetch http://dist.carla.org/carla/dists/focal/InRelease  Could not connect to [dist.carla.org:80](http://dist.carla.org/) (34.227.255.250), connection timed out
W: Some index files failed to download. They have been ignored, or old ones used instead.
```
Documentation I was following:
https://carla.readthedocs.io/en/0.9.15/start_quickstart/

mmezl, 2023-12-21:
Same issue for me...

nayuta-yanagisawa, 2023-12-22:
I've encountered the same issue. According to #6332, the server seems to be down for more than half a year.

mmezl, 2023-12-22:
Thanks for info. I am trying to follow instructions with comiling of files but stuck now with Unreal engine Editor.

QuteSaltyFish, 2024-01-02:
Got the same problem here.

wangziyi2016, 2024-01-03:
Same problem

johndahl, 2024-01-12:
Still the same problem.

extendedkalman, 2024-01-17:
Still the same problem

QingMounta, 2024-01-19:
same problem，It hasn't been fixed yet

DeepDuke, 2024-01-20:
same  problem...

DeepDuke, 2024-01-20:
@nsubiron @bernatx @Axel1092 @marcgpuig @DSantosO @manishthani @yaknostoyok @doterop @felipecode @corkyw10 @pravinblaze @iFuSiiOnzZ @glopezdiest @joel-mb @adrian-ollero @jackbart94 @algababr @hiteboar @dnovillo-CVC @juaxix @germanros1987 @fdomf @LuisPoveda @MarcelPiNacy @TheNihilisticRobot @berndgassmann @MattRoweEAIF @Blyron We need your help. I think it's a urgent problem. TKS!

Alionlion, 2024-01-22:
same issue...anyone tried 18.04? or it's just a common issue for Debian install?

haduoken, 2024-01-24:
same issue in 20.04

nenadilic84, 2024-01-24:
The server seems to be down for quite some time. I would suggest to add the manual instructions to main docs. Here are the steps:

#### 1. Install Required System Dependency
Before downloading CARLA, install the necessary system dependency:
```bash
sudo apt-get -y install libomp5
```

#### 2. Download the CARLA 0.9.15 Release
Download the CARLA_0.9.13.tar.gz file (approximately 16GB) from the official release:
```bash
wget https://carla-releases.s3.us-east-005.backblazeb2.com/Linux/CARLA_0.9.15.tar.gz
```

#### 3. Unpack CARLA to the Desired Directory
Unpack the downloaded file to `/opt/carla-simulator/`:
```bash
tar -xzvf CARLA_0.9.15.tar.gz -C /opt/carla-simulator/
```
#### 4. Install the CARLA Python Module
Finally, install the CARLA Python module and necessary dependencies:
```bash
python -m pip install carla==0.9.15
python -m pip install -r /opt/carla-simulator/PythonAPI/examples/requirements.txt
```

kamilkurach, 2024-02-10:
same issue (ubuntu 20.04)

ak-maker, 2024-09-10:
still cannot download from debian.

W: Failed to fetch https://dist.carla.org/carla/dists/jammy/InRelease  Temporary failure resolving 'dist.carla.org'

ParkJunBeom27, 2024-09-24:
In Ubuntu 20.04 can't connect to server now

brtmax, 2024-09-26:
Same issue

102hit, 2024-10-18:
> ```shell
> python -m pip install carla==0.9.15
> python -m pip install -r /opt/carla-simulator/PythonAPI/examples/requirements.txt
> ```

bro, did you solved it?

102hit, 2024-10-18:
> still cannot download from debian.
> 
> W: Failed to fetch https://dist.carla.org/carla/dists/jammy/InRelease Temporary failure resolving 'dist.carla.org'
> 
> On windows11, WSL2, ubuntu22.04

did you solve it ?

102hit, 2024-10-18:
> Same issue

did you solve it ?

qhaas, 2024-11-10:
The apt server is still down for Ubuntu 18.04 and 20.04

jihyeseo, 2025-01-17:
Still down. I tried from Ubuntu 22 and 24 inside UTM of M1.

```
Ign:85 http://dist.carla.org/carla jammy InRelease
Ign:85 http://dist.carla.org/carla jammy InRelease
Err:85 http://dist.carla.org/carla jammy InRelease
  Could not connect to dist.carla.org:80 (34.227.255.250), connection timed out
Fetched 30.4 MB in 37s (814 kB/s)
Reading package lists... Done
W: Failed to fetch http://dist.carla.org/carla/dists/jammy/InRelease  Could not connect to dist.carla.org:80 (34.227.255.250), connection timed out
W: Some index files failed to download. They have been ignored, or old ones used instead.
ubuntu@ubuntu:~$ 
```

QuteSaltyFish, 2025-01-17:
I think for now, the docker would be the better way to use carla

mcodinab, 2025-01-30:
This worked for me:
`nano /etc/apt/sources.list`

And then comment the line related with Carla server:
`deb [arch=amd64] http://dist.carla.org/carla focal main
`

Don't know the consequences of this but apt update then works.