carla-simulator/carla #7023: Vehicle doesn't follow the setting route

xinghanwen001, 2023-12-21:
I had set a route according to the spaw_points, and I want to achieve the goal that make the vehicle use the oppiste lane and drive for a while. However, when I set the autopilot mode, the vehicle itself could not follow the specified route and even turned around randomly.
Could you plase find a method to help me achieve the driving scene  "Using the opposite lane and back to the original lane" ?

Carla Version: 0.9.15
Platform: Windows

Codes:

def main():
    actor_list = []
    try:
        client = carla.Client(host='127.0.0.1', port=2000)
        client.set_timeout(2.0)
        world = client.get_world()
        blueprint_library = world.get_blueprint_library()
        spawn_points = world.get_map().get_spawn_points()
        # Setting route
        route_start = spawn_points[198]
        route_point = [150, 197, 94, 148, 194, 144]
        route = []
        for i in route_point:
            route.append(spawn_points[i].location)
            world.debug.draw_string(route_start.location, 'Start point', life_time=50, color=carla.Color(255, 0, 0))
        for ind in route_point:
            # spawn_points[ind].locations
            world.debug.draw_string(spawn_points[ind].location, str(ind), life_time=50, color=carla.Color(255, 0, 0))
        # Create Vehicle
        ego_vehicle_bp = blueprint_library.find('vehicle.tesla.model3')
        ego_vehicle_bp.set_attribute('color', '0,255,0')
        vehicle = world.try_spawn_actor(ego_vehicle_bp, route_start)
        vehicle.set_autopilot(True)

        actor_list.append(vehicle)
        while True:
            spectator = world.get_spectator()
            transform = vehicle.get_transform()
            spectator.set_transform(carla.Transform(transform.location + carla.Location(z=20),
                                                    carla.Rotation(pitch=-90)))

    finally:
        print("destroying actors")
        client.apply_batch([carla.command.DestroyActor(x) for x in actor_list])
        print("done")

[image]
[image]

ray9583, 2024-01-15:
The Agent Module allows you to configure the vehicle path. 
Convert the path coordinate list to a waypoint object and construct it in the form of a starting point-arrival point as a function of set_destination within basic_agent between coordinates. 
Simply put, the starting point arrival point is repeated between the coordinates.
There is a slight sense of expediency. 
Otherwise, create a tuple list of coordinates and road options to run the driving plan, 
and the module is in the Color/Navigation/Agent folder; 
road options are annoying to configure themselves.

xinghanwen001, 2024-01-16:
thank you for your kind help, I'll try.

dev3225, 2024-01-18:
Hello,
I am not able to run scripts using pygame window. could you tell me whether you are using any other port than 2000 to run them?

> thank you for your kind help, I'll try.

xinghanwen001, 2024-01-19:
> Hello, I am not able to run scripts using pygame window. could you tell me whether you are using any other port than 2000 to run them?
> 
> > thank you for your kind help, I'll try.

In my project, I always use port 2000.