carla-simulator/carla #7028: Vehicle agent not added to the crowd by some problem

lfxx, 2023-12-23:


lfxx, 2023-12-23:
> CARLA version:0.9.15
> 
> Platform/OS:Ubuntu20.04
> 
> Problem you have experienced: i try to add 700 Vehicles and 300 walkers in Town05,When i finished respawn Vehicles,then i need to respawn walkers with ai controller.I get the error above at this time.This led to inaccurate traffic behavior simulation, such as vehicles passing directly through pedestrians instead of stopping or colliding.
> 
> What you expected to happen: Vehicle should stop when walker is in front of it.
> 
> Steps to reproduce: just add 700 Vehicles then add 300 walkers in Town05.

This this very urgent.Any one come to help me out.Thanks very much!

mawistc, 2023-12-28:
I can not help directly, but just a hint, there are only 302 spawn points for vehicles available in Town05 - and maybe that is where the problem starts.

lfxx, 2024-01-02:
> I can not help directly, but just a hint, there are only 302 spawn points for vehicles available in Town05 - and maybe that is where the problem starts.

finally,it turns out that every map limit its respawned object to 500,if your object num > 500,you will get above error

closed 2024-01-02 by lfxx

qhaas, 2026-08-27:
Started seeing this in Town04 with only 90% of the spawn points being used for random vehicles, guessing not all the towns can handle multiple firetrucks.  The predetermined spawn points might need to be revisited given some of the cars have gotten bigger.