carla-simulator/carla #7029: Error using "automatic_control.py", module 'tensorflow.compat.v1' has no attribute 'contrib'.

SYS-Herr, 2023-12-25:


SYS-Herr, 2023-12-25:
CARLA version:0.9.13

Platform/OS:Ubuntu22.04

Problem you have experienced:
When I run "automatic_control.py" at line 61 (from agents.navigation.behavior_agent import BehaviorAgent) I get an error:
AttributeError: module 'tensorflow.compat.v1' has no attribute 'contrib'
Also, when I first finished installing Carla there was no error reported, however after I configured the interpreter with pycharm and chose to inherit the full site package I had this problem, thanks guys!
What you expected to happen:
error solution
Steps to reproduce:
I've tried uninstalling tensorflow and reinstalling it, but it seems that I can't install the 1.X version of the tensorflow library because of the python version issue
Also I'm not quite sure what happens if you choose to inherit the global site package when configuring the interpreter with pycharm, but I guess that's the source of the problem

ray9583, 2024-01-10:
That's right. If you need TensorFlow 1.x version to use Carla's agent module and you have Python 3.8 or later installed, installing TensorFlow 1.x version will be difficult. Instead of using TensorFlow 2 version, use Python 3.7 and TensorFlow 1.15.0 versions

hemozon, 2024-12-20:
Hello, I am also facing this issue. TensorFlow 1.x requires Python 3.7, but when I downgrade Python, CARLA becomes incompatible. Have you managed to resolve this?