carla-simulator/carla #7030: WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API  WARNING: Client API version     = 0.9.14  WARNING: Simulator API version  = 0.9.15-dirty

Xiaodiudiudiu, 2023-12-26:
Issue report:
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.14 
WARNING: Simulator API version  = 0.9.15-dirty 

Excuse me, I meet a bug :
[image]

I wonder how to figure out the incompatible issue, cause I just follow the official documents.
Could you please tell me how to find  "Simulator API Version = 0.9.14"? 

Thanks for your time and patience.

alexx-km, 2023-12-27:
I guess you build a custom version of Carla? In that case you'd also need to build and install the Python wheel for that version.

PatrickPromitzer, 2024-01-08:
The "Simulation API version" is the version of the carla you are running.
The "Client API Version" is the python egg or package you use in python.

If you didn't change something in the python API of carla, you can use
```comandline
path/to/env/bin/python -m pip install carla==0.9.15
```
The "-dirty" is just the name of the build you made with the git/source version.

You can use the *.egg or *.whl file in your carla build if you want the exact version that fit to your carla.
{CARLA_ROOT}/PythonAPI/carla/dist/

The *.egg is for the PYTHONPATH
```commandline
export path/to/carla-0.9.XXXXXX.egg
```
and the *.whl is for installing with pip

```comandline
path/to/env/bin/python -m pip install ppath/to/carla-0.9.XXXXXX.whl
```

xavisolesoft, 2024-04-26:
> The "Simulation API version" is the version of the carla you are running. The "Client API Version" is the python egg or package you use in python.
> 
> If you didn't change something in the python API of carla, you can use
> 
> ```
> path/to/env/bin/python -m pip install carla==0.9.15
> ```
> 
> The "-dirty" is just the name of the build you made with the git/source version.
> 
> You can use the *.egg or *.whl file in your carla build if you want the exact version that fit to your carla. {CARLA_ROOT}/PythonAPI/carla/dist/
> 
> The *.egg is for the PYTHONPATH
> 
> ```
> export path/to/carla-0.9.XXXXXX.egg
> ```
> 
> and the *.whl is for installing with pip
> 
> ```
> path/to/env/bin/python -m pip install ppath/to/carla-0.9.XXXXXX.whl
> ```

This is the fix, so I close this issue.

closed 2024-04-26 by xavisolesoft

hao-xiaoxing, 2024-10-21:
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.15 
WARNING: Simulator API version  = 0.9.15-228-geeb507e58 
这种算是版本统一了吗

yuhongsheng110, 2024-12-03:
@hao-xiaoxing 
应该不是统一， 我的情况和你类似：
场景：在ubuntu1804上运行examples，win10上运行Calar服务端，并设定好相应IP，Port,有如下警告：
WARNING: Client API version     = 0.9.15-228-geeb507e58 
WARNING: Simulator API version  = 0.9.15 
约5s后，example崩溃，服务端卡死；

EnsembleTesting, 2025-04-01:
@yuhongsheng110 @hao-xiaoxing 
Hi, I have the same issue. I run the CARLA server on Windows but PythonAPI (Client) on a Linux remote server. My mismatch is as follows:
WARNING: Version mismatch detected: You are trying to connect to a simulator that might be incompatible with this API 
WARNING: Client API version     = 0.9.15 
WARNING: Simulator API version  = 0.9.15-280-g7b0707548-dirty

I believe I cannot use my Windows .egg file to pip install carla in the Linux server. 

With this WARNING, I will meet some error when running scripts on Linux, for example:

>>> world.get_blueprint_library()
ERROR: invalid color autopilot 
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: role_name: colors must have 3 channels (R,G,B)

Note that I DO NOT have this "RGB Value Error" when I run "world.get_blueprint_library()" on Windows.

I think this is caused by the minor version mismatch.

Has anyone had any solutions yet?