carla-simulator/carla #7035: Lack of dir '/Game/Maps'

CMakey, 2023-12-27:
errors in treminal like this
`
[2023.12.27-12.07.30:673][  0]LogStreaming: Error: Couldn't find file for package /Game/Carla/Maps/Town01 requested by async loading code. NameToLoad: /Game/Carla/Maps/Town01
[2023.12.27-12.07.30:673][  0]LogStreaming: Error: Found 0 dependent packages...
[2023.12.27-12.07.31:682][  0]LogLinux: Error: appError called: Assertion failed:  [File:/var/lib/jenkins/UnrealEngine_4.18/Engine/Source/Runtime/CoreUObject/Private/Serialization/AsyncLoading.cpp] [Line: 6647] 
LoadPackageAsync failed to begin to load a package because the supplied package name was neither a valid long package name nor a filename of a map within a content folder: ''

Signal 11 caught.
Malloc Size=131076 LargeMemoryPoolOffset=131092 
`

Blyron, 2024-04-09:
In Unreal /Game/ Mean /Content/ 
Seems you haven't downloaded content properly

Blyron, 2024-04-19:
I am closing the issue, feel free to reopen

closed 2024-04-19 by Blyron