carla-simulator/carla #7039: Boost 1.80.0 cannot be downloaded

pradhanshrijal, 2024-01-01:
I had been running Carla on Ubuntu-22 for a while now. Since yesterday 31 Dec 2024, it seems like I cannot download the boost file anymore.

`Setup.sh: Extracting boost for Python 3.

gzip: stdin: not in gzip format
tar: Child returned status 1
tar: Error is not recoverable: exiting now
make: *** [Util/BuildTools/Linux.mk:137: setup] Error 2
`

When I run `file boost_1_80_0.tar.gz` on the file that is downloaded (and cannot be extracted), I get the following

`boost_1_80_0.tar.gz: HTML document, ASCII text, with very long lines (7078)`

I suspect the problem with the download of this file.
I tried to install 0.9.15 and 0.9.14, both of which was working just a few hours before.

tRNAoO, 2024-01-02:
I have the same error, and it seems like Boost hasn't been downloaded correctly. Have you resolved this issue? Additionally, I tried downloading Boost from https://archives.boost.io/release/1.80.0/source/, but this version requires clang-10, not clang-8.

tRNAoO, 2024-01-02:
> I have the same error, and it seems like Boost hasn't been downloaded correctly. Have you resolved this issue? Additionally, I tried downloading Boost from https://archives.boost.io/release/1.80.0/source/, but this version requires clang-10, not clang-8.

Using Carla 0.9.13 does not have this issue.

uehara-miraikan, 2024-01-05:
I met the same problem, and I commented out these lines in `Util/BuildTools/Setup.sh`

https://github.com/carla-simulator/carla/blob/d7b45c1e159e6d13296f7a3a4e8b13e6c2d62c18/Util/BuildTools/Setup.sh#L94-L99

and add
```
wget "https://carla-releases.s3.eu-west-3.amazonaws.com/Backup/${BOOST_PACKAGE_BASENAME}.tar.gz"
```

Then I can download boost correctly and continue the build process successfully.

pradhanshrijal, 2024-01-05:
> Using Carla 0.9.13 does not have this issue.

Hey, I need clang-12 for Ubuntu-22 so this did not work for me. But the downloads are working normally now.

pradhanshrijal, 2024-01-05:
> I met the same problem, and I commented out these lines in `Util/BuildTools/Setup.sh` [...]

Thanks for the comment, the download are working normally now, I will close this issue

pradhanshrijal, 2024-01-05:
Closing because the download link is working correctly now.

closed 2024-01-05 by pradhanshrijal

jannisjung, 2025-02-06:
> I have the same error, and it seems like Boost hasn't been downloaded correctly. Have you resolved this issue? Additionally, I tried downloading Boost from https://archives.boost.io/release/1.80.0/source/, but this version requires clang-10, not clang-8.

Thank you, for windows, in the file `carla\Util\InstallersWin\install_boost.bat` I have changed the line
```bat
set BOOST_REPO=https://boostorg.jfrog.io/artifactory/main/release/%BOOST_VERSION%/source/%BOOST_TEMP_FILE%
```
to
```bat
set BOOST_REPO=https://archives.boost.io/release/%BOOST_VERSION%/source/%BOOST_TEMP_FILE%
```
which made it work fine

pradhanshrijal, 2025-02-09:
https://pradhanshrijal.github.io/blog/carla-from-source/

Anonymi1ty, 2025-02-19:
> > I have the same error, and it seems like Boost hasn't been downloaded correctly. Have you resolved this issue? Additionally, I tried downloading Boost from https://archives.boost.io/release/1.80.0/source/, but this version requires clang-10, not clang-8.我有同样的错误，似乎Boost尚未正确下载。您解决了这个问题吗？此外，我尝试从https://archives.boost.io/release/1.80.0/source/下载Boost，但是此版本需要Clang-100，而不是Clang-8。
> 
> Thank you, for windows, in the file `carla\Util\InstallersWin\install_boost.bat` I have changed the line谢谢您的窗口，在文件中 `carla\Util\InstallersWin\install_boost.bat` 我改变了行
> 
> set BOOST_REPO=https://boostorg.jfrog.io/artifactory/main/release/%BOOST_VERSION%/source/%BOOST_TEMP_FILE%
> to  到
> 
> set BOOST_REPO=https://archives.boost.io/release/%BOOST_VERSION%/source/%BOOST_TEMP_FILE%
> which made it work fine这使其正常工作

For my case, I modified the line in the `carla/Util/BuildTools/Setup.sh` file from:

```
wget "https://boostorg.jfrog.io/artifactory/main/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" || true
```

to:

```
wget "https://archives.boost.io/release/${BOOST_VERSION}/source/${BOOST_PACKAGE_BASENAME}.tar.gz" || true
```