carla-simulator/carla #7045: multi lidar simulate and visualiza issue

tdcsu, 2024-01-04:
Hi, there,

I am trying to simulate a 16 line raycast lidar and a 64 raycast lidar on a vehicle, but there is a problem.
My script was adapted from CARLA-0.9.10/PythonAPI/examples/open3d_lidar.py
the whole code is below:
[script.txt](https://github.com/carla-simulator/carla/files/13825975/script.txt)

As you can see, to simulate two lidars, I just copied the existing lidar setup but changed some attributes.
However, when I start the CARLAUE4 and run my script, the visualise result is:
[image]

That means the second lidar works not well.

And when I stop the script, there said:
[image]
But when I only simulate one lidar, it works well.

Anyone can help me ?
Thanks with my appreciation

PatrickPromitzer, 2024-01-08:
The function you give the listener is in a subprocess, and Python makes a copy of the class/variables if you send them in a subprocess.
``` 
lidar1.listen(lambda data1: semantic_lidar_callback(data1, point_list1))
# semantic_lidar_callback() called in a subprocess
```

I can't say how o3d.geometry.PointCloud() works with the listener, but you can try using Queue.
Queue is made for handling data between the main-process and sub-process.

```python
from queue import Queue
data_queue = Queue()
...
# lidar
lidar1.listen(lambda data1: semantic_lidar_callback(data1, data_queue))

...
# in semantic_lidar_callback
# put data in the Queue
data_queue.put(point_cloud.raw_data)

...
# in while loop
# get data from the Queue
lidar_data = data_queue.get()
```

tdcsu, 2024-01-08:
您好，我已收到您的邮件，会尽快处理~

tdcsu, 2024-01-09:
> The function you give the listener is in a subprocess, and Python makes a copy of the class/variables if you send them in a subprocess.
>
> ```
> lidar1.listen(lambda data1: [...]
@PatrickPromitzer Thanks for your reply and your detailed suggestion. This issue has been solved. And I'll close this issue.

closed 2024-01-09 by tdcsu