carla-simulator/carla #7046: Collision Sensor is not working or getting segmentation fault

SExpert12, 2024-01-04:
Here is my code:
I am working on synchronous mode with multiple vehicles. Here is my code:
def main():
    try:
        client = carla.Client('localhost', 2000)
        client.set_timeout(5.0)
        world = client.get_world()
        points = get_vehloc(world)
        #print("points are:",points)
        batch = []
       
        start_time = 0.0
        end_time = 70.0
        delta_time = 0.05
        

        vehicle_bp = world.get_blueprint_library().filter('model3')[0]
        collision_sensor_bp = world.get_blueprint_library().find('sensor.other.collision')
        col_location = carla.Location(0,0,0)
        col_rotation= carla.Rotation(0,0,0)
        col_transform = carla.Transform(col_location,col_rotation)
            

       control= carla.VehicleControl(throttle = 0.4, steer = 0.0)
        MaxVeh = 6
        if MaxVeh > len(points):
            msg = 'requested %d vehicles, but could only find %d spawn points'
            logging.warning(msg, MaxVeh, len(points))
            exit()
         
       world.apply_settings(carla.WorldSettings(True,False,delta_time))
       #collision_sensor.listen(lambda collision_event:print("A collision"))   # not working 
       for n, transform in enumerate(points):
            vehicle = world.spawn_actor(vehicle_bp, transform)
            collision_sensor = world.spawn_actor(collision_sensor_bp, col_transform, attach_to=vehicle, 
                                attachment_type=carla.AttachmentType.Rigid)
            #collision_sensor.listen(lambda collision_event:print("A collision"))  # Segmentation fault

            vehicle.apply_control(carla.VehicleControl(throttle=0.35, steer=0))

        while start_time <= end_time:
           
            #collision_sensor.listen(lambda collision_event:print("A collision"))# core dump error
            world.tick(10)
            #collision_sensor.listen(lambda collision_event:print("A collision"))# core dump error
            world_snapshot = world.get_snapshot()
            #collision_sensor.listen(lambda collision_event:print("A collision"))# core dump error
                
            for actor_snapshot in world_snapshot:
                
                actual_actor = world.get_actor(actor_snapshot.id)
                if actual_actor.type_id == 'vehicle.tesla.model3':
                       veh_loc = actor_snapshot.get_transform().location
                       veh_roadid= world.get_map().get_waypoint(actor_snapshot.get_transform().location).road_id,
                       veh_sec = world.get_map().get_waypoint(actor_snapshot.get_transform().location).section_id
                      veh_road = veh_roadid[0]
             start_time +=0.05
            print("\n\nONE STEP IS OVER\nSimulation time is now", start_time)  

finally:
        world.apply_settings(carla.WorldSettings(False,False,0))
        if collision_sensor.is_alive:
            collision_sensor.destroy()
        print("simulation is over")

if __name__ == '__main__':
    try:
        main() 
    except KeyboardInterrupt:
        pass
    finally:
        print("\n done")

            
Where should I write the code where collision sensor listen the collision?
I am extracting information from world snapshot? 
Am I making any mistake in this code?

I want the information of the all the vehicles who are colliding and want to destroy them immediately as they are changing the route of other vehicles when they collide.

I have write in this code when l write code to listen to the collision sensor not working or getting error as segmentation fault.

I need help to resolve this issue.

Thanks in advance

Blyron, 2024-04-15:
This a code example for collision sensor, could you check if this is working for you? The attachment type is not necesary for sensors. After that could you check the logs? If you are not using a package they should be located in "CARLA_ROOT/Unreal/CarlaUE4/Saved/Logs"

```

from __future__ import print_function

import argparse
import glob
import math
import os
import sys
import time
try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass
try:
    sys.path.append(os.path.dirname(os.path.dirname(os.path.abspath(__file__))) + '/carla')
except IndexError:
    pass

import carla

def _on_collision(event):
    print(f"Collision with {event.other_actor.type_id}")

def main():

    client = carla.Client()
    world = client.get_world()
    bp_lib = world.get_blueprint_library()
    spectator = world.get_spectator()

    # Spawn the actor
    bp = bp_lib.filter("*walker*")[0]
    # bp.set_attribute('is_invincible', 'false')
    walker = world.spawn_actor(bp, carla.Transform(carla.Location(200.7, 199.3, 0.2), carla.Rotation()))

    bp = bp_lib.filter("*mkz_2020*")[0]
    vehicle = world.spawn_actor(bp, carla.Transform(carla.Location(177.7, 198.8, 0.2), carla.Rotation()))
    spectator.set_transform(carla.Transform(carla.Location(205.9, 193.2, 3.9), carla.Rotation(pitch=-29, yaw=135)))

    collision_bp = bp_lib.find('sensor.other.collision')
    sensor_collision = world.spawn_actor(collision_bp, carla.Transform(), attach_to=vehicle)
    sensor_collision.listen(lambda event: _on_collision(event))

    vehicle.apply_control(carla.VehicleControl(throttle=1))

    for _ in range(400):
        world.tick()

    walker.destroy()
    sensor_collision.destroy()
    vehicle.destroy()

if __name__ == '__main__':
    main()
```

Blyron, 2024-04-19:
We have updated dev branch, could you update your code and then try this again to know if that is working now?

Blyron, 2024-04-26:
Hello again, any updates? otherwise I will close this issue

SExpert12, 2024-04-30:
Okay. Thanks. Let me check this.

Blyron, 2024-05-21:
No but maybe someone does. Open a discussion or search for it in the discussions. I will be closing this issue for now.

closed 2024-05-21 by Blyron