carla-simulator/carla #7047: Errors concerning the `osm-world-renderer`, includes and dependencies

Zaynab1997, 2024-01-04:
Greetings,

I have downloaded the provided project and noticed that there is a project `osm-world-renderer` inside.
Launching it using the following : 
```console
./Run.sh
```
and encountered various errors, despite the attempts of fixing them given the execution of `MapDrawer.cpp`, even after installing and compiling `libosmscout` and `lunasvg` libraries, and fixing the paths and imports.

Can you provide a detailed tutorial of how to run this package?

matejm42, 2024-01-08:
It's used to import maps into Carla from .osm files. It's mentioned in documentation of [Digital Twin Tool](https://carla.readthedocs.io/en/latest/adv_digital_twin/). But it is not working for me

ZaynabEM, 2024-01-12:
@matejm42 Thank you for your response. 

Yes this is exactly what I was looking for, and the goal is to create a new map in carla for me. I was trying this method, with `blender` and [CADMAPPER](https://cadmapper.com/) for the buildings, as an alternative to the [RoadRunner](https://carla.readthedocs.io/en/latest/tuto_M_generate_map/) solution.

If I understand correctly, the dependencies `libosmscout` and `lunasvg` are correct, but I get the following : 
```console
Deleting existing build/ directory.
Creating build/ directory.
-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/User/carla/osm-world-renderer/build
[ 16%] Building CXX object CMakeFiles/OsmRenderer.dir/OsmRenderer/src/OsmRenderer.cpp.o
[ 33%] Building CXX object CMakeFiles/OsmRenderer.dir/OsmRenderer/src/MapDrawer.cpp.o
[ 50%] Building CXX object CMakeFiles/OsmRenderer.dir/OsmRenderer/src/MapRasterizer.cpp.o
[ 66%] Linking CXX static library libOsmRenderer.a
[ 66%] Built target OsmRenderer
[ 83%] Building CXX object CMakeFiles/osm-world-renderer.dir/OsmRenderer/main.cpp.o
[100%] Linking CXX executable osm-world-renderer
/usr/bin/ld: cannot find -losmscout: No such file or directory
/usr/bin/ld: cannot find -losmscout_map: No such file or directory
/usr/bin/ld: cannot find -losmscout_map_svg: No such file or directory
/usr/bin/ld: cannot find -llunasvg: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/osm-world-renderer.dir/build.make:98: osm-world-renderer] Error 1
make[1]: *** [CMakeFiles/Makefile2:111: CMakeFiles/osm-world-renderer.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
./Run.sh: line 20: ./osm-world-renderer: No such file or directory
```

I tried adding the path `.so` for each library to the `LD_LIBRARY_PATH`, but it didn't work as well.