carla-simulator/carla #7053: Carla Toold Content folder missing

matejm42, 2024-01-08:
CARLA version: 0.9.15
Platform/OS: Ubuntu 20.04
Problem you have experienced: In the `Content Browser` in `Unreal Engine Editor` is only folder *Content* and the *Carla Content*, *Carla C++ Classes*, *Carla Tools Content*, *HoudiniEngine Content*, *StreetMap Content* are missing.
[image]
What you expected to happen:
Steps to reproduce:
Other information (documentation you consulted, workarounds you tried):

matejm42, 2024-01-08:
I was not able to successfully use the Digital Twin Tool anyway, but that is already described in #6913

Sivagopinathreddy-Vinta, 2024-01-26:
in ContentBrowser Right clivk and select all views to see all folders in content view

leilai125, 2024-02-29:
I have the same issue. Have you solved it?

matejm42, 2024-03-01:
@leilai125 I "solved" it by creating the symlink to Plugins inside of the Content folder.

> I found UW_OnRoadMainWidget in Plugins folder and solved this issue by creating a symlink to this folder inside of Content folder I could see in the Content Browser
> I ran from the CALRA_ROOT/Unreal/CarlaUE4/Content Folder
> 
>```
>   ln -s /full/path/to/carla_root/Unreal/CarlaUE4/Plugins Plugins
>```

leilai125, 2024-03-01:
Hi,

Thanks! It works! Did you eventually generate a map from openstreetmap like the link (https://carla.org/2023/11/10/release-0.9.15/) shows? I am still struggling in generating the same result as the link shows.

matejm42, 2024-03-01:
@leilai125 I was able to generate some map.  I had to choose map without buildings, otherwise it crashes. It still threw errors and the product was not accurate :( 
https://github.com/carla-simulator/carla/issues/6913#issuecomment-1881058612

Blyron, 2024-04-19:
To see Plugins content you can 
UE5: https://www.youtube.com/watch?v=G3Q3I3-gH3A&ab_channel=Algosyntax
UE4: https://www.youtube.com/watch?v=05Ch89wQabk&ab_channel=AntonIovbak

If you have issues with other features please open other issue.

I am closing this issue. Please feel free to reopen it

closed 2024-04-19 by Blyron

Sephiroth1118, 2024-09-03:
Hi, I have the same problem but stranger. I even don't have the plugin folder in my content. My content is like this.
[image]
I tried to run Update.sh to update the content. But it told me content is already up-to-date.

SkatingReimu, 2024-09-18:
Hi all, this was a simple solution to my issue (I was trying to figure out to get the plugins to show up, not sure if you guys have tried this)
On the bottom right of the Content Browser, there should be a "View Options" menu that you can click on. Clicking on "Show Plugin Content" allowed me to access the folder. 
[image]