carla-simulator/carla #7061: make osmreneder error / fatal error: pango/pangoft2.h: No such file or directory

dbparkJ, 2024-01-12:
In file included from /home/geon_lab/carla/osm-world-renderer/OsmRenderer/src/MapDrawer.cpp:5:
/home/geon_lab/carla/osm-world-renderer/ThirdParties/include/osmscoutmapsvg/MapPainterSVG.h:34:12: fatal error: pango/pangoft2.h: No such file or directory
   34 |   #include <pango/pangoft2.h>
      |            ^~~~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/OsmRenderer.dir/build.make:90: CMakeFiles/OsmRenderer.dir/OsmRenderer/src/MapDrawer.cpp.o] Error 1
make[3]: Leaving directory '/home/geon_lab/carla/Build/osm-world-renderer-build'
make[2]: *** [CMakeFiles/Makefile2:85: CMakeFiles/OsmRenderer.dir/all] Error 2
make[2]: Leaving directory '/home/geon_lab/carla/Build/osm-world-renderer-build'
make[1]: *** [Makefile:91: all] Error 2
make[1]: Leaving directory '/home/geon_lab/carla/Build/osm-world-renderer-build'
make: *** [Util/BuildTools/Linux.mk:160: osmrenderer] Error 2

`make osmrender`
That is my error logs
plz help

jiakaa, 2024-01-14:
I re-add the missing dependencies and made it.....

dbparkJ, 2024-01-15:
@jiakaa 
Thank you for your response
But where can I find Dendency??
If you deleted and reinstalled an existing folder, the same error still appears

omerfarukadastec, 2024-01-16:
Hi,

I got the same error. You need to include some of libraries to the CMakeLists.txt into "osm-world-renderer".

Just go to the /carla/osm-world-renderer/CMakeLists.txt and add the lines below:

`include_directories(/usr/include/pango-1.0)
include_directories(/usr/include/freetype2)
include_directories(/usr/include/glib-2.0)
include_directories(/usr/lib/x86_64-linux-gnu/glib-2.0/include)
include_directories(/usr/include/harfbuzz)`

You should install these packages/libraries too.

dbparkJ, 2024-01-17:
@omerfarukadastec 
Thank you for your answer
As you said, the make osmender command was successfully executed when I added it to the Cmake file.

However, in the process of going to {CARLA_ROOT}/Build/libosmcout-source/maps/build.sh <path_to_osm_file>
./build.sh: 8: [[: not found
./build.sh: 10: [[: not found
./build.sh: 12: [[: not found
Cannot find Import executable!
An error like this has occurred. Can you help?

Just in case, I will attach my sample file
[sample.zip](https://github.com/carla-simulator/carla/files/13957749/sample.zip)

Aysenayilmaz, 2024-01-23:
> @omerfarukadastec Thank you for your answer As you said, the make osmender command was successfully executed when I added it to the Cmake file.
>
> However, in the process of going to [...]

I'm having the same problem. Did you find a solution?

omerfarukadastec, 2024-01-23:
Hi,

I didnt get this error you mentioned but I checked the build.sh. Your error is about this code block below:

```
if [[ -x ../Import/src/Import ]]; then
  importExe=../Import/src/Import
elif [[ -x ../debug/Import/Import ]]; then
  importExe=../debug/Import/Import
elif [[ -x ../build/Import/Import ]]; then
  importExe=../build/Import/Import
else
  echo "Cannot find Import executable!"
  exit 1
fi
```
"build.sh" script check the "Import.cpp" file in these three directories. You have to have "Import.cpp" file like below:

[image]

Can you check this file is here or not?

Aysenayilmaz, 2024-01-23:
> Hi,
> 
> I didnt get this error you mentioned but I checked the build.sh. Your error is about this code block below:
> 
> ```
> if [[ -x ../Import/src/Import ]]; then
>   importExe=../Import/src/Import
> elif [[ -x ../debug/Import/Import ]]; then
>   importExe=../debug/Import/Import
> elif [[ -x ../build/Import/Import ]]; then
>   importExe=../build/Import/Import
> else
>   echo "Cannot find Import executable!"
>   exit 1
> fi
> ```
> 
> "build.sh" script check the "Import.cpp" file in these three directories. You have to have "Import.cpp" file like below:
> 
> ![Screenshot from 2024-01-23 12-21-23](https://private-user-images.githubusercontent.com/104622368/298879923-ab5bd968-df05-477e-a9a6-ce071166189d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDYwMDIyOTIsIm5iZiI6MTcwNjAwMTk5MiwicGF0aCI6Ii8xMDQ2MjIzNjgvMjk4ODc5OTIzLWFiNWJkOTY4LWRmMDUtNDc3ZS1hOWE2LWNlMDcxMTY2MTg5ZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEyM1QwOTI2MzJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02NDdkNGI5NWI1YWY1YmIyMWUwMDM0YzA5ZDdmYjgzYTFjZjJmMDExYmVlYmNiMDM4OWNjNWFmNzNiOWUyMzEyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2Q4yDN2svyJu8JR21g_cozxww5B0-fVig8dE1hqNq9A)
> 
> Can you check this file is here or not?

I don't have that cpp file. I do not have the CarlaToolsContent/OnroadMapGenerator in these files, which is required to use osmrenderer with the URL. I guess there was a problem with the build.

omerfarukadastec, 2024-01-23:
Hi,

After a little bit tampering, I think I understand the issue. "build.sh" is checking the executable of Import, not the Import.cpp. So, we need to build it with CMake.

I went back to the "~/carla/Build/libosmscout-source/Import" and did
```
cmake ..
make
```

After ending build process, an executable file "Import" is created into "/home/adastec/carla/Build/libosmscout-source/Import/Import/" folder.

Also you should change the "build.sh" file like below:

```
if [[ -x ../Import/Import/Import ]]; then
  importExe=../Import/Import/Import
elif [[ -x ../debug/Import/Import ]]; then
  importExe=../debug/Import/Import
elif [[ -x ../build/Import/Import ]]; then
  importExe=../build/Import/Import
else
  echo "Cannot find Import executable!"
  exit 1
fi
```

After this change, build.sh works. Can you try it @dbparkJ?

@Aysenayilmaz , are you sure that the carla version is 0.9.15? If version is correct, maybe you can re-install CARLA. Also, do you work on Linux or Windows? I am working on Linux and there are no problem like yours. If you work on Linux, maybe you can run "make osmrenderer" again. You can follow with these link https://github.com/carla-simulator/carla/blob/master/Docs/adv_digital_twin.md

zjr-bit, 2024-02-25:
> Hi,
>
> After a little bit tampering, I think I understand the issue. "build.sh" is checking the executable of Import, not the Import.cpp. So, we need to build it with CMake.
>
> I went back to the [...]

Hi, execuse me, I would like to ask when you used the digital twin tool, did you meet some bugs, such as failing to build maps which invovles buildings.  Thanks for your information.

omerfarukadastec, 2024-02-26:
> Hi, execuse me, I would like to ask when you used the digital twin tool, did you meet some bugs, such as failing to build maps which invovles buildings. Thanks for your information.

Hi, you can look at this issue #6913

zjr-bit, 2024-02-26:
> Hi, you can look at this issue #6913

Hi, thanks for your reply. I have met the same problem. But after I tried many times, I made a map successfully. The .osm file contains several buildings, but in Carla there is no buildings. Unluckily, I don't find any hints.