carla-simulator/carla #7062: Can I use Carla 0.9.15 with leaderboard2.0?

nemonswx, 2024-01-12:
As far as I know, Carla 0.9.15 is newer comparing to leaderboard-2.0:
1. For Carla it self, there is 0.9.15 build and leaderboard-2.0 specific build;
2. For leaderboard repo, there is only a leaderboard2.0 branch that is up to date;
3. For scenario_runner repo, there are both master branch and leaderboard-2.0 branch that are up to date and they have diverted for quite a lot of commits.
I would like to take full advantage of latest Carla0.9.15 features along with leaderboard2.0 routes and metrics, so can I use Carla0.9.15 build with leaderboard-2.0 branch for leaderboard repo and master branch for scenario_runner repo?

PabloVD, 2024-05-17:
In principle the expected CARLA package to be used in the Leaderboard is the one [in the documentation](https://leaderboard.carla.org/get_started/), which is 0.9.14 with some bug fixes, but you should not have problems with using 0.9.15.

NayaBaslan, 2024-10-17:
Did this issue get fixed?
Does Leaderboard-2.0 work on Carla 0.9.15 ?

PabloVD, 2024-11-08:
While the Leaderboard 2.0 was designed to work with 0.9.14 as stated in the documentation, it should not present problems with 0.9.15.

closed 2024-11-08 by PabloVD