carla-simulator/carla #7064: ~/carla/PythonAPI/carla/dependencies/lib/libproj.a(geodesic.c.o): relocation R_X86_64_32S against `.rodata.cst8' can not be used when making a shared object; recompile with -fPI

jiakaa, 2024-01-13:
**CARLA version:** 0.9.15 lastest
**Platform/OS:** ubuntu 20.4
**Problem I have experienced:** 
I encountered an error while executing the "`make PythonAPI`" command in a line of command “`/usr/bin/env python3 setup.py bdist_egg bdist_wheel --dist-dir dist/.tmp`” in the script "`BuildPythonAPI.sh`", and I have been unable to resolve it. **Please help**!
```python
BuildPythonAPI.sh: Building Python API for Python 3.
+ [[ -z '' ]]
+ /usr/bin/env python3 setup.py bdist_egg bdist_wheel --dist-dir dist/.tmp
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
writing source/carla.egg-info/PKG-INFO
writing dependency_links to source/carla.egg-info/dependency_links.txt
writing top-level names to source/carla.egg-info/top_level.txt
reading manifest file 'source/carla.egg-info/SOURCES.txt'
writing manifest file 'source/carla.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_py
running build_ext
building 'carla.libcarla' extension
/home/zou_yj/UnrealEngine_4.26/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Idependencies/include -I/usr/include/python3.8 -c source/libcarla/libcarla.cpp -o build/temp.linux-x86_64-3.8/source/libcarla/libcarla.o -isystem dependencies/include/system -fPIC -std=c++14 -Werror -Wall -Wextra -Wpedantic -Wno-self-assign-overloaded -Wdeprecated -Wno-shadow -Wuninitialized -Wunreachable-code -Wpessimizing-move -Wold-style-cast -Wnull-dereference -Wduplicate-enum -Wnon-virtual-dtor -Wheader-hygiene -Wconversion -Wfloat-overflow-conversion -DBOOST_ERROR_CODE_HEADER_ONLY -DLIBCARLA_WITH_PYTHON_SUPPORT -stdlib=libstdc++ -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-x86_64-3.8/source/libcarla/libcarla.o -Ldependencies/lib -o build/lib.linux-x86_64-3.8/carla/libcarla.cpython-38-x86_64-linux-gnu.so /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libcarla_client.a /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/librpc.a /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libboost_filesystem.a /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libRecast.a /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libDetour.a /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libDetourCrowd.a /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libosm2odr.a /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libxerces-c.a -lz /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libproj.a /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libsqlite3.a /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libboost_python38.a -lpng -ljpeg -ltiff -lstdc++
/usr/bin/ld: /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libproj.a(geodesic.c.o): relocation R_X86_64_32S against `.rodata.cst8' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libproj.a(wkt1_generated_parser.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: /home/zou_yj/carla/PythonAPI/carla/dependencies/lib/libproj.a(wkt2_generated_parser.c.o): relocation R_X86_64_32S against `.rodata' can not be used when making a shared object; recompile with -fPIC
collect2: error: ld returned 1 exit status
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
make: *** [Util/BuildTools/Linux.mk:92：PythonAPI] 错误 1
```

jiakaa, 2024-01-13:
I solved this problem by deleting all the cache files and reinstalling it, and succeeded in compiling it.

closed 2024-01-13 by jiakaa

Croquembouche, 2024-01-31:
@jiakaa 
Can you clarify on deleting the cache files? cmake cache?

thank you

nebulaJinyoujun, 2024-02-27:
I'm having the same issue as you. I'm trying 0.9.14 right now. Did you solve this problem?

nebulaJinyoujun, 2024-02-27:
according to this issue https://github.com/carla-simulator/carla/issues/6474, first rm Build/proj-install, then add -DCMAKE_C_FLAG="-fPIC" to Setup.sh, the part of compile PROJ. At last, make clean , make PythonAPI

Croquembouche, 2024-03-14:
I have found the solution. Delete the proj file from the CARLA Build Folder. Then rerun the carla make launch script to download and re-install.

Seems like a problem with unzipping the file and building the document.

Similar solution to #6474