carla-simulator/carla #7067: where can I find the https://github.com/CarlaUnreal/UnrealEngine.git

zczjx, 2024-01-14:
in the [https://carla.readthedocs.io/en/0.9.15/build_linux/](0.9.15 Linux build)
we need the  https://github.com/CarlaUnreal/UnrealEngine.git for the next step libCarla build
then can we build pythonAPI for python 3.8

but where can we find the lib, without it, we cannot build the next steps

PatrickPromitzer, 2024-01-15:
You need to link an Unreal Engine's account linked to your GitHub account.

> 
> Unreal Engine
> 
> Starting with version 0.9.12, CARLA uses a modified fork of Unreal Engine 4.26. This fork contains patches specific to CARLA.
> 
> **Be aware that to download this fork of Unreal Engine, you need to have a GitHub account linked to Unreal Engine's account.** If you don't have this set up, please follow [this guide](https://www.unrealengine.com/en-US/ue4-on-github) before going any further.
> 
Here the link to the guid.
[https://www.unrealengine.com/en-US/ue-on-github](https://www.unrealengine.com/en-US/ue-on-github)

zczjx, 2024-01-15:
let me try, thanks @PatrickPromitzer

zczjx, 2024-01-23:
verified and closed

closed 2024-01-23 by zczjx