carla-simulator/carla #7069: make PythonAPI error : Could not find compiler set in environment variable CC

GZ0KyLee, 2024-01-15:


nusstu0019, 2024-01-18:
I solve this problem by reinstall.

viktor-from, 2024-02-07:
> I solve this problem by reinstall.

What did you reinstall? Clang or the Carla repo?

nusstu0019, 2024-02-07:
carla repo

        - External Email -

yuxinhi, 2024-03-27:
I'm facing the same difficulty. Have you resolved this problem ?

arashfree, 2024-04-03:
I was forgotten to set the correct $UE4_ROOT, set it correct so it would be solved. it should be your installed Unreal folder.

LuisPoveda, 2024-04-12:
Is this issue been resolved?
Are you fine with closing it?

LuisPoveda, 2024-04-19:
Since there is no response I am going to close this issue, feel free to reopen it.

closed 2024-04-19 by LuisPoveda

MOHAMMEDHUSSEIN94, 2024-04-29:
> > I solve this problem by reinstall.
> 
> What did you reinstall? Clang or the Carla repo?
> 
> Edit: Seems like my export UE4_ROOT=~/UnrealEngine_4.26 variable in .profile just wasn't set correctly.

I did so and used echo $UE4_ROOT to make sure the directory is that of unreal engine and still getting the same error

MOHAMMEDHUSSEIN94, 2024-04-29:
> I was forgotten to set the correct $UE4_ROOT, set it correct so it would be solved. it should be your installed Unreal folder.

I did so using gedit ~/.bashrc ,wrote[export UE4_ROOT=~/UnrealEngineCarla] at the end and then ran [source ~/.bashrc] and used echo $UE4_ROOT to make sure the directory is that of unreal engine and still getting the same error

YuvarajP, 2024-05-28:
I am getting the same error

dragnil1, 2024-06-24:
I was facing this error while using `sudo make launch`. Firstly, I tried using `make launch`. But got **permission denied** error as my `carla` folder did not have sufficient permission. Then, I enabled read and write permission using the accepted answer of this [question](https://askubuntu.com/questions/487527/give-specific-user-permission-to-write-to-a-folder-using-w-notation). After that, `make launch` worked nicely. Let me know if solution is not appropiate. I also followed [this](https://stackoverflow.com/questions/31421327/cmake-cxx-compiler-broken-while-compiling-with-cmake).

JiachengZuo, 2024-07-25:
Maybe your clang version is too old; I solved this problem by using these command.
`sudo apt update 
 sudo apt install clang`

OYan60718, 2024-07-26:
> > I was forgotten to set the correct $UE4_ROOT, set it correct so it would be solved. it should be your installed Unreal folder.
> 
> I did so by writing at the end of bashrc [export UE4_ROOT=~/UnrealEngineCarla] and then ran source and used echo $UE4_ROOT to make sure the directory is that of unreal engine and still getting the same error ![image](https://private-user-images.githubusercontent.com/25697041/326606374-b61be7fc-e4aa-4d20-a2ac-3947abc2ea85.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjE5Njk0MjgsIm5iZiI6MTcyMTk2OTEyOCwicGF0aCI6Ii8yNTY5NzA0MS8zMjY2MDYzNzQtYjYxYmU3ZmMtZTRhYS00ZDIwLWEyYWMtMzk0N2FiYzJlYTg1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA3MjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNzI2VDA0NDUyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE0ODgwOWY1NzViN2FkOTQyMDU5OTY1YTFjMTA4YjFkNTgxOWYyYzgwNmViNTVhNDc2OWRhMjQwN2FmM2ViMGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.kpcWVWNKgKCVaGUNfPwZB-fYdLmU8zRtyRojwqu7OSw)

You need to close and reopen the terminal after setting the path

OYan60718, 2024-07-26:
> Maybe your clang version is too old; I solved this problem by using these command. `sudo apt update ` ` sudo apt install clang`

Do not try this, this will install the latest version of clang, which is not what CARLA requires.

latercoder, 2024-09-06:
> I was facing this error while using `sudo make launch`. Firstly, I tried using `make launch`. But got **permission denied** error as my `carla` folder did not have sufficient permission. Then, I [...]

Works for my case!
For some reason, I compiled Carla Client with command "sudo make PythonAPI", then I came across the error "Could not find compiler set in environment variable CC...", then I reinstalled carla from step "git clone ...", and compiled with command "make PythonAPI", then fixed. If you know how to change the permission of carla folder, just try as @dragnil1 did.

raynbowy23, 2024-10-01:
I spent the whole day trying to make it work. Basically, every past answer sounds acceptable but does not specify enough.

If you encounter this type of error, it means it does not correctly link to your installed Unreal Engine for some reason.
```/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang```

The correct path should put a path to the Unreal Engine at the front that you are supposed to install already, according to the Carla build document.
```$UE4_ROOT/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang```.

I tried to set env path in my .bashrc and .bash_profile but the program didn't read it for some reason. Instead, I manually modified the build files and added path, which is UE4_ROOT in front of CC, the path for clang.

In my case specifically, I wrote as below to `Setup.sh`, `BuildOSM2ODR.sh`, `BuidPythonAPI.sh` (maybe I forgot some), which all under the path `/opt/carla/Util/BuildTools/`.
```
export UE4_ROOT="/home/YOUR USERNAME/UnrealEngine_4.26"
export CC="$UE4_ROOT/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v17_clang-10.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang"
```

Find CC path and add UE4_ROOT. Hope it helps!

linitau, 2025-04-05:
it's wrong if you set the UE environment path as "export UE4_ROOT=~/UnrealEngine_4.26", you should replace "~" as absolute path