carla-simulator/carla #7073: Windows 10: make PythonAPI stops at BuildOSM2ODR. CMake Error: The source directory "D:/carla/Build/osm2odr-visualstudio/x64" does not exist.

juanrodriguezdekra, 2024-01-16:
Hi,

I am trying to build the Carla 0.9.15 latest source code on Windows 10.

When I do make PythonAPI in Visual studio 19 x64 command prompt, the process stops
at BuildOSM2ODR with CMAKE error. I got this error:

CMake Error: The source directory "D:/carla/Build/osm2odr-visualstudio/x64" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
Error: could not load cache
El sistema no puede encontrar el archivo especificado.
El sistema no puede encontrar el archivo especificado.
-[BuildOSM2ODR]: OSM2ODR has been successfully installed in "D:\carla\PythonAPI\carla\dependencies\"
-[BuildPythonAPI]: [Batch params]: --py3
Building Python API for Python 3.
compiling:
  - source/libcarla/libcarla.cpp
running bdist_egg
running egg_info
creating source\carla.egg-info
writing source\carla.egg-info\PKG-INFO
writing dependency_links to source\carla.egg-info\dependency_links.txt
writing top-level names to source\carla.egg-info\top_level.txt
writing manifest file 'source\carla.egg-info\SOURCES.txt'
reading manifest file 'source\carla.egg-info\SOURCES.txt'
writing manifest file 'source\carla.egg-info\SOURCES.txt'
installing library code to build\bdist.win-amd64\egg
C:\Users\dekra\AppData\Roaming\Python\Python312\site-packages\setuptools\_distutils\cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
running install_lib
running build_py
creating build
creating build\lib.win-amd64-cpython-312
creating build\lib.win-amd64-cpython-312\carla
copying source\carla\command.py -> build\lib.win-amd64-cpython-312\carla
copying source\carla\__init__.py -> build\lib.win-amd64-cpython-312\carla
running build_ext
building 'carla.libcarla' extension
creating build\temp.win-amd64-cpython-312
creating build\temp.win-amd64-cpython-312\Release
creating build\temp.win-amd64-cpython-312\Release\source
creating build\temp.win-amd64-cpython-312\Release\source\libcarla
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\bin\HostX86\x64\cl.exe" /c /nologo /O2 /W3 /GL /DNDEBUG /MD -Idependencies/include "-IC:\Program Files\Python312\include" "-IC:\Program Files\Python312\Include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\ATLMFC\include" "-IC:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30133\include" "-IC:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\include\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\ucrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\shared" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\um" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\winrt" "-IC:\Program Files (x86)\Windows Kits\10\include\10.0.22000.0\cppwinrt" /EHsc /Tpsource/libcarla/libcarla.cpp /Fobuild\temp.win-amd64-cpython-312\Release\source/libcarla/libcarla.obj /experimental:external /external:W0 /external:I dependencies/include/system /DBOOST_ALL_NO_LIB /DBOOST_PYTHON_STATIC_LIB /DBOOST_ERROR_CODE_HEADER_ONLY /D_WIN32_WINNT=0x0600 /DHAVE_SNPRINTF /DLIBCARLA_WITH_PYTHON_SUPPORT -DLIBCARLA_IMAGE_WITH_PNG_SUPPORT=true /MD
libcarla.cpp
D:\carla\PythonAPI\carla\source\libcarla\OSM2ODR.cpp(7): fatal error C1083: No se puede abrir el archivo incluir: 'OSM2ODR.h': No such file or directory
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x64\\cl.exe' failed with exit code 2

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\carla\PythonAPI\carla\dist"!

When I look into OSM2ODR.bat file it should be in line 112 like:

cmake -G %GENERATOR% %PLATFORM%^

But instead of that, it seems that my OSM2ODR.bat file has been build in a different way, I don,t know why. 
This is my OSM2ODR.bat file:

#! /bin/bash
DOC_STRING="Build OSM2ODR."

[... 123 lines cut ...]
  ninja osm2odr
  ninja install

  mkdir -p ${OSM2ODR_SERVER_BUILD_FOLDER}
  cd ${OSM2ODR_SERVER_BUILD_FOLDER}

  LLVM_BASENAME=llvm-8.0
  LLVM_INCLUDE="$UE4_ROOT/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1"
  LLVM_LIBPATH="$UE4_ROOT/Engine/Source/ThirdParty/Linux/LibCxx/lib/Linux/x86_64-unknown-linux-gnu"

  cmake ${OSM2ODR_SOURCE_FOLDER} \
      -G "Eclipse CDT4 - Ninja" \
      -DCMAKE_CXX_FLAGS="-fPIC -std=c++14 -stdlib=libc++ -I${LLVM_INCLUDE} -L${LLVM_LIBPATH}" \
      -DCMAKE_INSTALL_PREFIX=${LIBCARLA_INSTALL_SERVER_FOLDER} \
      -DPROJ_INCLUDE_DIR=${CARLA_BUILD_FOLDER}/proj-install-server/include \
      -DPROJ_LIBRARY=${CARLA_BUILD_FOLDER}/proj-install-server/lib/libproj.a \
      -DXercesC_INCLUDE_DIR=${CARLA_BUILD_FOLDER}/xerces-c-3.2.3-install-server/include \
      -DXercesC_LIBRARY=${CARLA_BUILD_FOLDER}/xerces-c-3.2.3-install-server/lib/libxerces-c.a

  ninja osm2odr
  ninja install

fi

log " OSM2ODR Success!"

pseventekidis, 2024-01-19:
Try to delete the whole carla directory and clone from git again. 
I had the same issue when I downloaded manually the source code. 
With git clone from command line it was gone

juanrodriguezdekra, 2024-01-22:
I,ve done and it works. I then proceeded to change the command on line 112 to:
cmake -G %GENERATOR% -A x64^

In the PythonAPI make everything was correct and without incident. 
[image]

But in the make launch I got make: *** [CarlaUE4Editor] Error 6. Apparently without a fatal error, as you can see in the attached image.
[image]

 This is the new point I'm stuck at, in case anyone knows how to get out of here.

pseventekidis, 2024-01-22:
I will share my experience once more for the windows build. 
All problems for me were due to the fact that I downloaded the code manually and did not use the git command. 
Maybe you can also delete the carla directory completely and try again (provided Unreal is ok )

juanrodriguezdekra, 2024-01-24:
#6362

juanrodriguezdekra, 2024-01-24:
Check this issue #6898 .

closed 2024-01-24 by juanrodriguezdekra