carla-simulator/carla #7079: flying/floating vehicles in Town12 Carla 9.15

Fornerio, 2024-01-18:
Thanks for contributing to CARLA!

If you are reporting an issue, please use the following outline:

CARLA version: 0.9.14 and 0.9.15
Platform/OS:  Linux and Windows

Hello everyone, I am using CARLA map **Town12** to collect some vehicular data while driving on a highway among with other cars. What happens is that in a very specific area of the map, like there was an **invisible obstacle**, all the non-hero vehicles start flying or floating, often creating accidents and generating unrealistic behaviours, resulting in distorted vehicle data.

I have tried using both **9.15** and **9.14**, and I have no clue about how to solve this problem. Could it be the fault of the map itself? Does anyone ever faced this issue? 

Here a video of the problem:
https://github.com/carla-simulator/carla/assets/24523864/d5ff6705-8068-41e3-9199-584702025923

Thank you in advance to all of you,

Regards,
Luca

ray9583, 2024-01-20:
I've never witnessed this before. Is this only happening in town12?

Fornerio, 2024-01-22:
Hi @ray9583 , As far as I know this is happening in Town12 only. However, since I need a large map for my experiments I discarded the possibility to use a smaller one, but I will try simulating also on different maps and see whether the problem also occurs in other towns

dwkang02, 2024-07-02:
I am also experiencing the same issue. It occurs at a specific location. Is there a solution?
<position x="547.2" y="6093.4" z="360.2"/>
<position x="453.3" y="6134.9" z="358.6"/>
<position x="367.0" y="6190.7" z="358.0"/>