carla-simulator/carla #7080: Change of simulation speed in specific Town12 zones

Fornerio, 2024-01-18:
CARLA version: 0.9.14 and 0.9.15
Platform/OS: Linux and Windows

Hello everyone, I am using CARLA map **Town12** to collect some vehicular data while driving on a highway among with other cars. In certain areas of the map, the **simulation speed changes**, causing inconsistency. To collect data through multiple simulations with different users' using a steering wheel, it is necessary for the simulation speed to remain constant regardless of the map and its areas, otherwise the driving experience would be ruined.

Despite trying both **9.15** and **9.14**, I am unsure how to solve this problem. The cause may be the rendering of certain objects, but this is uncertain to me. Has anyone else experienced this issue?

Here a video of the problem:

https://github.com/carla-simulator/carla/assets/24523864/d2d0cc88-9907-4e94-846c-7603699a2a2a

If you look at the fps and pay attention to the simulation, you will notice that these change whenever the car turns back, but this is not the only one portion of map where this happens. In certain sections of the highway, changing direction does not necessarily trigger this occurrence; it can happen simply by driving straight ahead.

Thank you in advance to all of you,

Regards,
Luca

chirudeep001, 2024-01-19:
Processor:	Intel(R) Xeon(R) w5-2465X   3.10 GHz
Installed RAM:  64.0 GB (63.5 GB usable)
GPU : Nvidia Geforce RTX 4080
OS: windows

i have high end processor and GPU eventhough i am not able to acheive higher FPS(scenario runner) and i have seen you have deployed 174 vehicles and you are in range of 30 fps.

Fornerio, 2024-01-19:
> Processor: Intel(R) Xeon(R) w5-2465X 3.10 GHz Installed RAM: 64.0 GB (63.5 GB usable) GPU : Nvidia Geforce RTX 4080 OS: windows
> 
> i have high end processor and GPU eventhough i am not able to acheive higher FPS(scenario runner) and i have seen you have deployed 174 vehicles and you are in range of 30 fps.
> 
> My CPU and GPU usage does not even cross 50%.

Hello,
I don't think that this problem is related to GPU or the whole computer since I tried with both higher and lower-end computers. It's just a problem that occurs in certain position regardless of the computer. Anyway, I edited my first comment and added the info about the computer I used for that very specific simulation (but the very same problem, with lower fps, occurs on my laptop as well)

chirudeep001, 2024-01-19:
Hi @Fornerio thank you for the reply, so the problem is with the server side because when i dont use the scenario runner i get very higher FPS and i am ubale to deploy almost 100 walkers and 100 vehicles, but the fps problem occurs only when i use scenario runner.

are you using pygame window as client and server as carla or running carla in server window if so can you help me out.

Thank you

Fornerio, 2024-01-19:
Hi @chirudeep001, I don't use carla scenario runner, I just drive along the map using `manual controller.py`. I use pygame as client and I run carla as server. 

Regarding the comparison between Ubuntu and Windows, right now I am using my laptop (I'm abroad and I can't use my workspace) with Windows installed. The laptop is definitely less performing, but the same problems occur in the same pattern, which is why I think it is not a PC problem. I also checked the carla ticks, but everything seems to be OK.

chirudeep001, 2024-01-19:
Than you

Madecu, 2024-04-19:
Hi @Fornerio 
First of all, we apologize for the delay in responding to the issues. We have been investing the majority of our resources in other tasks. However, we will now return to providing more support to the community. We apologize for any inconvenience this may have caused.

The FPS drops in some areas of Town12 becaus the maps is divided in several tiles. So the fps drop is caused becaus in that point you are probably spawning a new chunk

closed 2024-05-03 by Madecu