carla-simulator/carla #7082: Windows 11: make PythonAPI and has error: LINK : fatal error LNK1104: cannot open file 'libboost_filesystem-vc143-mt-x64-1_80.lib' error: command 'D:\\Visual Studio 2022\\VC\\Tools\\MSVC\\14.38.33130\\bin\\HostX86\\x64\\link.exe' failed with exit code 1104

aminoed, 2024-01-19:
Hi. When I was trying to give the command make PythonAPI on Win11, it said Carla lib for python has been successfully installed in "D:\carla\PythonAPI\carla\dist", but the dist folder doesn't exist at all.
Some parts of the make process is shown below:

> -- The C compiler identification is MSVC 19.38.33134.0
> -- The CXX compiler identification is MSVC 19.38.33134.0
> -- Detecting C compiler ABI info
> -- Detecting C compiler ABI info - done
> -- Check for working C compiler: D:/Visual Studio 2022/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe - skipped-- Detecting C compile features
> -- Detecting C compile features - done
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Check for working CXX compiler: D:/Visual Studio 2022/VC/Tools/MSVC/14.38.33130/bin/Hostx64/x64/cl.exe - skipped
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> CMake Deprecation Warning at CMakeLists.txt:35 (cmake_minimum_required):
>   Compatibility with CMake < 3.5 will be removed from a future version of
>   CMake.
> 
>   Update the VERSION argument <min> value or use a ...<max> suffix to tell
>   CMake that the project does not need compatibility with older versions.
> 
> 
> -- CMAKE_BINARY_DIR: D:/carla/Build/osm2odr-visualstudio
> -- CMAKE_SOURCE_DIR: D:/carla/Build/om2odr-source
> --
> -- Platform:
> --     Host: Windows-10.0.22621 AMD64
> --     Target: Windows-10.0.22621 AMD64
> --     CMake: 3.27.9
> --     CMake generator: Visual Studio 17 2022
> --     CMake build tool: D:/Visual Studio 2022/MSBuild/Current/Bin/amd64/MSBuild.exe
> --     Compiler: MSVC 19.38.33134.0
> --
> -- Found XercesC: D:\carla\Build\\xerces-c-3.2.3-install\lib\xerces-c.lib (found version "3.2.3")
> -- Found Proj: D:/carla/Build/proj-install/lib/proj.lib
> -- Found ZLIB: D:/carla/PythonAPI/carla/dependencies/lib/zlib.lib (found version "1.2.13")
> -- Enabled features: Windows-10.0.22621 AMD64 MSVC 19.38.33134.0 Release Proj
> -- Configuring done (6.5s)
> -- Generating done (0.0s)
> -- Build files have been written to: D:/carla/Build/osm2odr-visualstudio
> MSBuild version 17.8.5+b5265ef37 for .NET Framework
> 
>   1>Checking Build System
>   Building Custom Rule D:/carla/Build/om2odr-source/src/CMakeLists.txt
>   OSM2ODR.cpp
>   NBAlgorithms.cpp
>   NBAlgorithms_Railway.cpp
>   NBAlgorithms_Ramps.cpp
>   NBConnection.cpp
>   NBContHelper.cpp
>   NBDistrict.cpp
>   NBDistrictCont.cpp
>   NBEdge.cpp
>   NBEdgeCont.cpp
>   NBFrame.cpp
>   NBHeightMapper.cpp
>   NBHelpers.cpp
>   NBLoadedSUMOTLDef.cpp
>   NBLoadedTLDef.cpp
>   NBNetBuilder.cpp
>   NBNode.cpp
>   NBNodeCont.cpp
>   NBNodeShapeComputer.cpp
>   NBOwnTLDef.cpp
>   NBPTLine.cpp
>   NBPTLineCont.cpp
>   NBPTPlatform.cpp
>   NBPTStop.cpp
>   NBPTStopCont.cpp
>   NBParking.cpp
[... 288 lines cut ...]
>         ]
> dependencies/include/system\boost/iterator/iterator_facade.hpp(550): note: see the first reference to 'boost::gil::dereference_iterator_adaptor<Iterator,Deref>::dereference' in 'boost::iterators::iterator_core_access::dereference'
>         with
>         [
>             Iterator=boost::gil::bgra8_ptr_t,
>             Deref=boost::gil::color_convert_deref_fn<const boost::gil::pixel<uint8_t,boost::gil::bgra_layout_t> &,DstPixelT,carla::image::ColorConverter::Depth>
>         ]
> dependencies/include/system\boost/gil/image_view_factory.hpp(144): note: see reference to function template instantiation 'void carla::image::ColorConverter::Depth::operator ()<const _Ty,DstP>(const SrcPixelT &,DstPixelT &) const' being compiled
>         with
>         [
>             _Ty=boost::gil::bgra8c_pixel_t,
>             DstP=DstPixelT,
>             SrcPixelT=boost::gil::bgra8c_pixel_t,
>             DstPixelT=DstPixelT
>         ]
> dependencies/include\carla/image/ColorConverter.h(35): warning C4244: 'initializing': conversion from 'int' to 'const float', possible loss of data
> D:\Visual Studio 2022\VC\Tools\MSVC\14.38.33130\bin\HostX86\x64\link.exe /nologo /INCREMENTAL:NO /LTCG /DLL /MANIFEST:EMBED,ID=2 /MANIFESTUAC:NO /LIBPATH:dependencies/lib /LIBPATH:D:\Python\libs /LIBPATH:D:\Python\PCbuild\amd64 /LIBPATH:D:\Visual Studio 2022\VC\Tools\MSVC\14.38.33130\ATLMFC\lib\x64 /LIBPATH:D:\Visual Studio 2022\VC\Tools\MSVC\14.38.33130\lib\x64 /LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x64 /LIBPATH:D:\Windows Kits\10\lib\10.0.22621.0\ucrt\x64 /LIBPATH:D:\Windows Kits\10\\lib\10.0.22621.0\\um\x64 /LIBPATH:D:\Visual Studio 2022\VC\Tools\MSVC\14.38.33130\ATLMFC\lib\x64 /LIBPATH:D:\Visual Studio 2022\VC\Tools\MSVC\14.38.33130\lib\x64 /LIBPATH:C:\Program Files (x86)\Windows Kits\NETFXSDK\4.8\lib\um\x64 /LIBPATH:D:\Windows Kits\10\lib\10.0.22621.0\ucrt\x64 /LIBPATH:D:\Windows Kits\10\\lib\10.0.22621.0\\um\x64 /EXPORT:PyInit_libcarla build\temp.win-amd64-3.9\Release\source/libcarla/libcarla.obj /OUT:build\lib.win-amd64-3.9\carla\libcarla.cp39-win_amd64.pyd /IMPLIB:build\temp.win-amd64-3.9\Release\source/libcarla\libcarla.cp39-win_amd64.lib shlwapi.lib Advapi32.lib ole32.lib shell32.lib D:\carla\PythonAPI\carla\dependencies/lib\carla_client.lib D:\carla\PythonAPI\carla\dependencies/lib\Detour.lib D:\carla\PythonAPI\carla\dependencies/lib\DetourCrowd.lib D:\carla\PythonAPI\carla\dependencies/lib\libboost_filesystem-vc142-mt-x64-1_80.lib D:\carla\PythonAPI\carla\dependencies/lib\libboost_python39-vc142-mt-x64-1_80.lib D:\carla\PythonAPI\carla\dependencies/lib\libpng.lib D:\carla\PythonAPI\carla\dependencies/lib\osm2odr.lib D:\carla\PythonAPI\carla\dependencies/lib\proj.lib D:\carla\PythonAPI\carla\dependencies/lib\Recast.lib D:\carla\PythonAPI\carla\dependencies/lib\rpc.lib D:\carla\PythonAPI\carla\dependencies/lib\sqlite3.lib D:\carla\PythonAPI\carla\dependencies/lib\xerces-c_3.lib D:\carla\PythonAPI\carla\dependencies/lib\zlib.lib
> LINK : fatal error LNK1104: cannot open file 'libboost_filesystem-vc143-mt-x64-1_80.lib'
> error: command 'D:\\Visual Studio 2022\\VC\\Tools\\MSVC\\14.38.33130\\bin\\HostX86\\x64\\link.exe' failed with exit code 1104
> 
> -[BuildPythonAPI]: Carla lib for python has been successfully installed in "D:\carla\PythonAPI\carla\dist"!

And two of the main error I got is :
> LINK : fatal error LNK1104: cannot open file 'libboost_filesystem-vc143-mt-x64-1_80.lib'
> error: command 'D:\\Visual Studio 2022\\VC\\Tools\\MSVC\\14.38.33130\\bin\\HostX86\\x64\\link.exe' failed with exit code 1104

OYan60718, 2024-03-30:
For this error, try to rebuild Carla with VS2019, as vs2022 uses msvc 14.3 while the toolset used to build boost is 14.2. Otherwise, try to install boost with 14.3 manually should also work, but not tested.

Zhenlin-Xu, 2024-04-11:
Sadly, visual studio 2019 community version is not available now

Zhenlin-Xu, 2024-04-11:
Solved with VS 2022!

As @OYan60718 suggested, first install boost with msvc 14.3.

```
C:\Users\LocalAdmin\Develop\carla>Util\InstallersWin\install_boost.bat --build-dir "C:\Users\LocalAdmin\Develop\carla\Build\" --toolset msvc-14.3 --version 1.80.0 -j 32
```

Then, rerun the 
```
make PythonAPI GENERATOR="Visual Studio 17 2022"
````

Finally, install the wheel file
```
py -m pip install PythonAPI\carla\dist\carla-0.9.15-cp37-cp37m-win_amd64.whl
```

It works for me! Thanks for your suggestions, OYan60718!

OYan60718, 2024-04-11:
> Sadly, visual studio 2019 community version is not available now
> 
> ![image](https://private-user-images.githubusercontent.com/86121429/321565559-fe1b5d75-d305-49b4-bde5-0bdcc6187dc9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI4Nzg5MDgsIm5iZiI6MTcxMjg3ODYwOCwicGF0aCI6Ii84NjEyMTQyOS8zMjE1NjU1NTktZmUxYjVkNzUtZDMwNS00OWI0LWJkZTUtMGJkY2M2MTg3ZGM5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDExVDIzMzY0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdkNDMzNDFkMWMxNDc4MGYyMmU3NDViOGY0NzQ5NjhmNjkyZjE0ZTVlM2QzMDBhNGE3YzZkMTQ2YmZiNzg2ZTcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.hSXw493brGC92SygmqMBZrCAIaF8kVlb9FywpU5XaPw)
> 
> Anybody has a solution? Thanks in advance!

Glad to hear you solved it. However, VS2019 is still available, there is a link provided on CARLA's build on Windows webpage if anyone is having trouble finding it.

zqy22, 2024-05-05:
> Solved with VS 2022!
>
> As @OYan60718 suggested, first install boost with msvc 14.3.
>
> ```
> C:\Users\LocalAdmin\Develop\carla>Util\InstallersWin\install_boost.bat --build-dir [...]

Thanks a lot. It works!

epochDVKHN, 2025-03-13:
> Glad to hear you solved it. However, VS2019 is still available, there is a link provided on CARLA's build on Windows webpage if anyone is having trouble finding it.

That link now works only for those with a VS Subscription. I found this one instead, hosted on Wayback Machine: [VS 2019 Community Edition](https://web.archive.org/web/20211009045628/https://download.visualstudio.microsoft.com/download/pr/1051e775-b2c9-4b7a-a227-1e60bffe102a/c758f79e86d619d6d1998fd67820f4970d803c28f447f503acc183df003719ec/vs_Community.exe).

_(Source: [Medium](https://medium.com/@ailene.chan/the-struggle-is-real-installing-dreyevr-carla-0-9-13-d68d0d1cd625))_

M-Colley, 2026-10-26:
Draft PR #10024 addresses the false-success/error-propagation part of Windows Python API builds on `ue4-dev`. It does not supply the missing Boost library or resolve its underlying configuration. The draft has batch regressions and currently has merge conflicts; a complete build remains unrun.