carla-simulator/carla #7089: Offset in big vehicle Bounding Boxes position

AitorIglesias, 2024-01-24:
CARLA version: 9.14
Platform/OS: Linux
Problem you have experienced: There is an offset in the position of the bounding box of big vehicles.

I am currently working to generate auto-labeled data. I generate point clouds, images and bounding boxes. I get the bounding boxes for each actor and I noticed that the bounding box for large vehicles like trucks or buses does not perfectly match the object in the point cloud. It can be seen in these images:

[image]
[image]

I know it is not the code because this does not happen with other actors or environment objects, as you can see in the example:

[image]
[image]

yyxssm, 2024-01-26:
Same question to me, how to solve it?

gsmario, 2024-05-17:
Hi, I had a look at the bus, and you're right, but there is a reason:
If you look at this image, you can see how the left mirror is creating that offset, since the right mirror is not in the same position as the left. So when creating the bounding box, unreal is adding that extra size to the left side of the bus.
The bounding box is working as intended.
[image]

gsmario, 2024-05-17:
I've also checked the other big vehicles, and as you can see, their bounding box is correct, so, which ones do you mean are wrong, besides the bus?
[image]

gsmario, 2024-05-24:
I'm closing the issue, since the problem seems solved.
Thanks for your time.

closed 2024-05-24 by gsmario