carla-simulator/carla #7092: AIWalkerController makes Pedestrians invisible in Lidar in 0.9.15

Pyrestone, 2024-01-25:
CARLA version: 0.9.15
Platform/OS: Linux Ubuntu 20.04
Problem you have experienced: Pedestrians are invisible in Lidar when an AIWalkerController is attached to them since .
What you expected to happen: Pedestrians are always visible in Lidar
Steps to reproduce: 
* spawn a pedestrian with an AIWalkerController controlling their behavior
* Spawn a Semantic Lidar Sensor on the ego vehicle
* Check the lidar pointcloud, and see that the pedestrian is not visible.

See also #6911 for a related issue with walkers becoming invisible in recordings.

All of this works fine with no problems in 0.9.14, but is broken since 0.9.15.

This might be related to "Optimizations" alluded to in the changelog regarding the collision mesh not being loaded as, but I can't test that. Either way, the issue was already confirmed by other users in Issue #6911

AreopagX, 2024-03-07:
@Pyrestone you're right. I took a look into the commit where these changes are mentioned and found that they disabled the actor physics and collisions for walkers. I assume that disabled physics causes the lidar rays to move through the walkers. I found two ways how you can solve this issue:

1. Using Python code
This way you don't have to rebuild the Carla client library. Just insert this piece of code after [starting the WalkerAIControllers](https://github.com/carla-simulator/carla/blob/60bd026b4822b4edb8a68cc17b9119866f303853/PythonAPI/examples/generate_traffic.py#L331C1-L337C71)
``` Python
for actor in all_actors:
  if "walker.pedestrian." in actor.type_id:
    actor.set_collisions(True)
    actor.set_simulate_physics(True)
```

2. Edit WalkerAIController.cpp
Remove these [three lines](https://github.com/carla-simulator/carla/blob/73315d3d62ad4bc9878f29ed2ac0faec79436a3e/LibCarla/source/carla/client/WalkerAIController.cpp#L26C1-L28C65) and rebuild the Carla client library. Maybe this undoes the alleged performance improvements, though you can see walkers in the lidar again :)

LuisPovedaCano, 2024-05-31:
This has been fixed, update the dev branch, feel free to reopen it if it occurs again

closed 2024-05-31 by LuisPovedaCano