carla-simulator/carla #7103: GPU Memory release

Mariusmarten, 2024-01-31:
I observed the following problem: Carla does not release the GPU 'freed' by destroying sensors.

**CARLA version**: 9.14

**Platform/OS**: carlasim/carla:0.9.14 docker image

**Problem**: GPU memory usage keeps on increasing. Once spawned sensors continue to occupy GPU space after being destroyed, the space is partially freed up before the GPU is almost full but this leads still to the data generation crashing at some point.

**Expected**: The GPU is directly freed up when the sensor is destroyed.

**Steps to reproduce**: Create and destroy sensors multiple times (e.g., spawn 24 camera sensors and destroy them 100 times in a loop).

**Other information**: I could not find further information about others experiencing similar problems or whether this is potentially expected behavior.

MarcelPiNacy, 2024-03-01:
Hi,
What system are you on?

Mariusmarten, 2024-03-11:
Hi, I am on a Linux based server system in a docker build on carlasim/carla:0.9.14 using this whl carla-0.9.14-cp38-cp38-manylinux_2_27_x86_64.whl

MarcelPiNacy, 2024-04-24:
Hi again, and apologies for the delay.
Is there any chance you could share the entire crash log?

Blyron, 2024-11-06:
This should be fixed in latest versions of ue4-dev branch

closed 2025-04-23 by Blyron