carla-simulator/carla #7121: How to Ignore traffic light with no slow down when approaching at intersection?

tkddnjs98, 2024-02-05:


tkddnjs98, 2024-02-05:
Hi. I'm trying to create a signalless intersection through a Carla simulator. Through the Carla Traffic manager, "ignore_traffic_lights" and "set_state, set_green_time," the vehicles always kept the green signal while ignoring the signal. However, there was always a case of slowing down whenever the vehicles entered the intersection and I want to make sure they kept their original speed when they entered the intersection. Is there any code that supports this? I'd appreciate it if you could let me know. 
I`m now using 0.9.13 and 0.9.15.
Additionally, previous versions of 0.9.11 had no corresponding issues.

https://github.com/carla-simulator/carla/assets/89904421/925ec3a6-32af-44af-920e-8b6d88792173

David-Burton-tcd, 2024-07-27:
I would also like to see some sort of answer to this. Did you ever figure it out?

MislavMores, 2025-01-08:
Hello, can you please tell me how did you manage to remove traffic lights?