carla-simulator/carla #7139: Attaching RGB camera to traffic lights in CARLA 0.9.15.

edvinborovac, 2024-02-09:
I am trying to attach 4 cameras to traffic lights on the same junction, 1 for each traffic light and collect data from those cameras. Each camera should be pointed to the lane that is coming towards that traffic light. How do I locate the spawnpoints and set those cameras?

AinaRoca, 2024-05-02:
I would suggest you to place the cameras manually and adjust them the way you want, then use the translation and rotation information you will find in details tab of each camera and use it to locate the spawnpoint of the camera.
Hope this helps, if that's not exactly what you asked for, let me know.

AinaRoca, 2024-05-17:
I consider this issue solved, so I'm going to close it.
If there is any doubt or problem, feel free to open it again.

closed 2024-05-17 by AinaRoca