carla-simulator/carla #7156: sensor.camera always cause memory leak since version 0.9.14 and 0.9.15

lygbuaa, 2024-02-18:
My carla server runs in official docker image, carlasim/carla:0.9.14 or carlasim/carla:0.9.15 (ubuntu 18.04), and my carla client runs in another docker (ubuntu 20.04). While carlasim/carla:0.9.13 never suffered this problem.
When I spawn a sensor.camera, the **sensor.camera.listen always cause memory leak** in CarlaUE4 server, while other sensors, like **sensor.lidar, won't cause memory leak**.
I found the similar issue, https://github.com/carla-simulator/carla/issues/6130, but no solution.
So anyone solved this problem?

LuisPoveda, 2024-04-12:
We are going to look at this

attr0, 2024-05-07:
Noticed the same issue on the windows version.

Blyron, 2024-11-06:
This should be fixed now. There shouldn't be more memory leak in listen functions. Try latest versions of Unreal 0.9.15

chuanye-Wang, 2024-11-16:
> This should be fixed now. There shouldn't be more memory leak in listen functions. Try latest versions of Unreal 0.9.15

I encountered the same issue:
···sh
/data/yangwenxian/project/simulator/carla/CarlaUE4.sh -RenderOffScreen -nosound -carla-rpc-port=22518 -graphicsadapter=6 None
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 136]
Exception thrown: bind: Address already in use [system:98 at /home/jenkins/workspace/carla_0.9.15/Build/boost-1.80.0-c10-install/include/boost/asio/detail/reactive_socket_service.hpp:161:33 in function 'bind']
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554
CommonUnixCrashHandler: Signal=11
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```
This is a custom map, which Ive export from a carlaUE4 editor. I create about 16 custom maps and import them into carla0.9.15(release version), some maps will have this "segmentation fault", some are not. Is there any solution? relly need help, Thanks.

Blyron, 2025-02-19:
This issue has been fixed in ue4-dev branch time ago

closed 2025-02-19 by Blyron