carla-simulator/carla #7160: Windows compilation failed during make PythonAPI

LarsWeber123, 2024-02-19:
Hi,

I am in the process of building Carla on Windows using the manual "https://carla.readthedocs.io/en/latest/build_windows/"

Platform/OS: Windows 10, I have installed VS2022 with all packages according to the build_windows manual

CARLA version: downloaded everything today just starting with Carla, not sure where to find the current Carla version number. I downloaded UE last week Thursday from the git, it compiled flawlessly and the Editor works, says version is 4.26.2.

Today I got errors at this point: 

`make PythonAPI GENERATOR="Visual Studio 17 2022"`

led to 
...failed updating 27 targets...
...skipped 4 targets...
...updated 16561 targets...

 -[install_boost]: [B2 ERROR] An error ocurred while installing using "b2.exe".
    -[install_boost]: Exiting with error...
scrolling up in the console output (x64 native tools command prompt) there are lots of occurrences "pyconfig.h No such file or directory"

First occurrence of this error can be found here:
...on 16500th target...
common.copy c:\Carla\carla\Build\boost-1.80.0-install\lib\cmake\boost_atomic-1.80.0\libboost_atomic-variant-vc142-mt-x64-1_80-static.cmake
build\boost\bin.v2\libs\atomic\build\msvc-14.2\release\link-static\threading-multi\libboost_atomic-variant-vc142-mt-x64-1_80-static.cmake
        1 Datei(en) kopiert.
compile-c-c++ build\boost\bin.v2\libs\filesystem\build\msvc-14.2\release\link-static\threading-multi\utf8_codecvt_facet.obj
utf8_codecvt_facet.cpp
boost-install.generate-cmake-config- build\boost\bin.v2\libs\python\build\install\boost_python-config.cmake
compile-c-c++ build\boost\bin.v2\libs\python\build\msvc-14.2\release\link-static\python-3.8\threading-multi\list.obj
list.cpp
.\boost/python/detail/wrap_python.hpp(57): fatal error C1083: Datei (Include) kann nicht geöffnet werden: "pyconfig.h": No such file or directory

There is approx. 20+ of these error messages afterwards. I have Python 3.11.5 installed on my Windows, installation path is "C:\Program Files\python3-11-5" which is in my Windows environment variables for my user. I have no other Python version installed right now, and there is no leftover stuff from previous Python installations mentioned in any other user / system environment variable. 

I have looked up the boost_python-config.cmake file mentioned above in the error output and it seems that Boost (by the way smake script says Boost 1.80.0) comes with its own Python ("boost_python_DIR" can be found in the cmake file). Maybe there is something mixed up here? Any ideas?

regards
Lars

LarsWeber123, 2024-02-20:
I figured it out, looking a bit more into the specific Boost - Python build issues that exist for ~10 years now :D  Based on what I found for this issue I added the following to "C:\Carla\carla\Build\boost-1.80.0-source\project-config.jam"

import python ;

if ! [ python.configured ]
{
    using python 
    : 3.11
    : "C:/python/python"
    : "C:/python/include"
    : "C:/python/libs"
    ;
}

closed 2024-02-20 by LarsWeber123

Zhenlin-Xu, 2024-04-11:
I would recommend you using python 3.7 when installing CARLA. It works for me when using VS 2022.

Just check with `py -V` to see if it is python 3.7.0