carla-simulator/carla #7163: Large Map (Town15) Camera Rendering Issue

GoodarzMehr, 2024-02-20:
CARLA version: 0.9.15
Platform/OS: Ubuntu 22.04
Problem you have experienced: far objects do not appear in camera (RGB/segmentation) images. For instance, an overhead (BEV) camera at 125m above the vehicle shows parked cars, but one at 1000m does not.
What you expected to happen: cameras (RGB/segmentation) should show far objects.
Steps to reproduce: 1. Spawn a vehicle with an overhead camera pointed downward. 2. Visualize the output for different distances (e.g. using OpenCV).

GoodarzMehr, 2024-02-21:
Example:

[image]

left: BEV camera 125m above the vehicle
right: BEV camera 1000m above the vehicle

Madecu, 2024-04-19:
Hi @GoodarzMehr 
First of all, we apologize for the delay in responding to the issues. We have been investing the majority of our resources in other tasks. However, we will now return to providing more support to the community. We apologize for any inconvenience this may have caused.

Town11 and higher are very large maps with many assets. For performance reasons, we need to apply techniques such as LOD or distance culling to reduce the number of polygons to be rendered based on the camera distance.

You can modify the distance at which objects disappear by adjusting the CullDistanceVolume in the map inside Unreal Engine. The rendering distance can be adjusted based on the size of the objects; larger objects have a determined greater distance, etc.

[image]

closed 2024-05-03 by Madecu