carla-simulator/carla #7164: VK_ERROR_OUT_OF_DEVICE_MEMORY crash when launching remotely via XRDP

muety, 2024-02-20:
I'm getting the following error when attempting to run CARLA on a machine that I'm connecting to via XRDP remote desktop: 

```
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
LowLevelFatalError [File:Unknown] [Line: 803] 
Result failed, VkResult=-2
 at /home/jenkins/UnrealEngine_4.26/Engine/Source/Runtime/VulkanRHI/Private/VulkanSwapChain.cpp:530 
 with error VK_ERROR_OUT_OF_DEVICE_MEMORY
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```

Starting CARLA while physically sitting in front of the screen works just fine. Any ideas?

Probably related to https://github.com/lgsvl/simulator/issues/1479. Perhaps not much that could be done on CARLA's end? 

**Specs**
- CARLA version: 0.9.15
- Platform/OS: Ubuntu 22.04
- xrdp version: 0.9.17

MarcelPiNacy, 2024-03-01:
Hi, could you provide the full log? (Or is that it?)

muety, 2024-03-01:
That's all, unfortunately.

milk-bottle-liyu, 2024-03-14:
Same problem with the same os and version. The device memory is 32G and the GPU memory is 8G which should be enough for Carla. Please help us with the problem~

torpedlars, 2024-05-10:
Encountered this error when i updated the GPU drivers, works on Nvidia 470 but get this error when using the 550 driver. Could solve it by reverting to the 470 driver. Not sure why this solves the error. Not sure if it helps you guys out.

muety, 2024-05-27:
Turns out I'm seeing the exact same error when logging into a Wayland session (as opposed to X11), even "in-person" (no remote desktop).