carla-simulator/carla #7165: Carla Server Crashes with TrafficLight

csonthomisi, 2024-02-20:
### Carla Server Crashes if TrafficLight is placed
CARLA version: **0.9.14 (Unreal Editor & Package)**
Platform/OS: **Ubuntu 22.04**
Problem you have experienced: **Can not start simulation with TrafficLights**
What you expected to happen: **The server crashes if TrafficLight is placed in the Map.**
Steps to reproduce: **Custom map imported from RoadRunner, place TrafficLight and start simulation**
Other information (documentation you consulted, workarounds you tried): **[Tried to place the traffic lights like written here](https://carla.readthedocs.io/en/0.9.14/tuto_content_authoring_maps/)**

I made a custom map, which was working fine.
I placed 2 Traffic Lights and exported the map package. After that i can not run the simulation neither in the UnrealEditor, and not from a packaged launcher. I tried 2 different types of TrafficLights (Ameriacan_Lights and BP_Traffic_Light) but none of it works.
I got the errors:
```
4.26.2-0+++UE4+Release-4.26 522 0
Disabling core dumps.
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=131160 LargeMemoryPoolOffset=196744 
Malloc Size=131160 LargeMemoryPoolOffset=327928 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```
```
LoginId:aa40adf836b14e63a5058740c516ca6e-000003e9
EpicAccountId:

Caught signal 11 Segmentation fault

libUE4Editor-Carla.so!UTrafficLightComponent::GetGroup() const [/media/mcsontho/Data/Carla_Build/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Traffic/TrafficLightComponent.cpp:145]
libUE4Editor-Carla.so!FWorldObserver::BroadcastTick(UCarlaEpisode const&, float, bool, bool) [/media/mcsontho/Data/Carla_Build/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor/WorldObserver.cpp:373]
libUE4Editor-Carla.so!FCarlaEngine::OnPostTick(UWorld*, ELevelTick, float) [/media/mcsontho/Data/Carla_Build/carla/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaEngine.cpp:327]
libUE4Editor-Carla.so!TBaseRawMethodDelegateInstance<false, FCarlaEngine, void (UWorld*, ELevelTick, float), FDefaultDelegateUserPolicy>::ExecuteIfSafe(UWorld*, ELevelTick, float) const [/media/mcsontho/Data/Carla_Build/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Delegates/DelegateInstancesImpl.h:467]
libUE4Editor-Engine.so!TMulticastDelegate<void (UWorld*, ELevelTick, float), FDefaultDelegateUserPolicy>::Broadcast(UWorld*, ELevelTick, float) const [/media/mcsontho/Data/Carla_Build/UnrealEngine_4.26/Engine/Source/Runtime/Core/Public/Delegates/DelegateSignatureImpl.inl:955]
libUE4Editor-Engine.so!UWorld::Tick(ELevelTick, float) [/media/mcsontho/Data/Carla_Build/UnrealEngine_4.26/Engine/Source/Runtime/Engine/Private/LevelTick.cpp:1593]
libUE4Editor-UnrealEd.so!UEditorEngine::Tick(float, bool) [/media/mcsontho/Data/Carla_Build/UnrealEngine_4.26/Engine/Source/Editor/UnrealEd/Private/EditorEngine.cpp:1716]
libUE4Editor-UnrealEd.so!UUnrealEdEngine::Tick(float, bool) [/media/mcsontho/Data/Carla_Build/UnrealEngine_4.26/Engine/Source/Editor/UnrealEd/Private/UnrealEdEngine.cpp:423]
UE4Editor!FEngineLoop::Tick() [/media/mcsontho/Data/Carla_Build/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4830]
UE4Editor!GuardedMain(char16_t const*) [/media/mcsontho/Data/Carla_Build/UnrealEngine_4.26/Engine/Source/Runtime/Launch/Private/Launch.cpp:171]
libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/media/mcsontho/Data/Carla_Build/UnrealEngine_4.26/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
libc.so.6!UnknownFunction(0x29d8f)
libc.so.6!__libc_start_main(+0x7f)
UE4Editor!_start()
```

DanielSioldea, 2024-07-16:
Had this same issue happen to me. Did you ever figure out what was causing it to crash? I've placed one traffic light to test and I cant run the server with it present.

chuanye-Wang, 2024-07-17:
Same issue, Did you figure out what happened?

chuanye-Wang, 2024-07-17:
This error happened in carla 0.9.15-dev too.

DanielSioldea, 2024-07-17:
I managed to make it work by creating a junction of multiple lights following the steps here: https://carla.readthedocs.io/en/latest/tuto_M_custom_add_tl/

Not sure if the issue is just happening when placing down a single light, but that's what worked for me.

chuanye-Wang, 2024-07-17:
It's fine while dagging the trafficlights into the scene, but as you click the play button in UE4edditor, you will get a crash report.

bty, I followed the same guide as you did

chuanye-Wang, 2024-07-17:
I ll try, thanks for that

chuanye-Wang, 2024-07-18:
@DanielSioldea Thanks! you're right.