carla-simulator/carla #7169: Vehicle Drives in Circles When I Apply a PID Controller

knowaiser, 2024-02-21:
CARLA version: 0.9.15
Platform/OS: Windows 10 Pro

Hello,

I am currently experiencing an issue with my project using the CARLA simulator. Despite my best efforts to implement a PID controller for vehicle navigation, the vehicle behaves unexpectedly by continuously driving in circles. The vehicle's steering and throttle seem to always hit their maximum signals, which seems to be a key factor in the circular driving pattern. I have experimented with adjusting the PID coefficients (Kp, Ki, Kd) but have seen little to no improvement in the vehicle's behavior.

PID Implementation: Utilized both a custom PID code and the VehiclePIDController class provided by CARLA. My implementation mainly differs in the logic used for waypoint generation.

I am including a video of the vehicle's performance to provide a clearer understanding of the issue at hand. 

I would greatly appreciate any insights, suggestions, or guidance on how to resolve this issue. Any feedback on the PID settings, waypoint generation, or other aspects of my implementation would be invaluable.

Thank you in advance for your help and support.

P.S. The blue dots are the waypoints from previous episodes.

https://github.com/carla-simulator/carla/assets/5973546/b3a84740-b24c-48d4-80a6-956705b2134d

SExpert12, 2024-03-04:
Can you please share your code? This happens to me multiple times.

MarcelPiNacy, 2024-05-03:
Hi,

Sorry for the late reply, could you share both the code and console output from this?

Marcel

PabloVD, 2024-05-17:
Without seeing the code we cannot know which could be wrong with the implementation.
[Here](https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/local_planner.py) you can see how the local planner can be employed to call the vehicle PID controller and  apply the control commands, and [here](https://github.com/carla-simulator/carla/blob/master/PythonAPI/carla/agents/navigation/basic_agent.py) you can see an application on a simple example.