carla-simulator/carla #7177: Launching CARLA server (./CarlaUE4.sh) without rendering at all.

yozaii, 2024-02-22:
Hello,

I am trying to run CARLA 0.9.15 on an Ubuntu 22.04 PC with very limited dedicated GPU VRAM (512Mb). Launching the CARLA server with ./CarlaUE4.sh never works, it starts a black screen then the process gets killed / my PC crashes, which is logical given the limited video memory. I want to use it in tandem with "python no_rendering_mode.py", but I first need an instance of the server to be launched.

I have already tried ./CarlaUE4.sh -RenderOffscreen, or with low quality, but I still have not been able to launch a successful instance of the server. Is there any workaround possible that avoids the usage of the GPU for rendering, or am I forced to upgrade my GPU as the only solution?

Thanks in advance

csonthomihaly, 2024-02-22:
Maybe try launching with `./CarlaUE4.sh -nullrhi `

yozaii, 2024-02-23:
> Maybe try launching with `./CarlaUE4.sh -nullrhi `

Exactly what I was looking for, thank you!

closed 2024-02-23 by yozaii