carla-simulator/carla #7179: facing issue of "libcarla.obj : error LNK2001: unresolved external symbol"

2002atulyadav, 2024-02-23:
I have tried to build the carla_latest, carla_0.9.14, carla_0.9.12.
Everytime i faced the same issue while running the **make PythonAPI**

error:
libcarla.obj : error LNK2001: unresolved external symbol "public: __thiscall carla::client::Map::Map(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >)" (??0Map@client@carla@@QAE@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@0@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: class std::vector<class boost::shared_ptr<class carla::client::Waypoint>,class std::allocator<class boost::shared_ptr<class carla::client::Waypoint> > > __thiscall carla::client::TrafficLight::GetStopWaypoints(void)const " (?GetStopWaypoints@TrafficLight@client@carla@@QBE?AV?$vector@V?$shared_ptr@VWaypoint@client@carla@@@boost@@V?$allocator@V?$shared_ptr@VWaypoint@client@carla@@@boost@@@std@@@std@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __thiscall carla::client::TrafficLight::GetOpenDRIVEID(void)const " (?GetOpenDRIVEID@TrafficLight@client@carla@@QBE?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: class std::vector<class carla::geom::BoundingBox,class std::allocator<class carla::geom::BoundingBox> > __thiscall carla::client::TrafficLight::GetLightBoxes(void)const " (?GetLightBoxes@TrafficLight@client@carla@@QBE?AV?$vector@VBoundingBox@geom@carla@@V?$allocator@VBoundingBox@geom@carla@@@std@@@std@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: class std::vector<class boost::shared_ptr<class carla::client::Waypoint>,class std::allocator<class boost::shared_ptr<class carla::client::Waypoint> > > __thiscall carla::client::TrafficLight::GetAffectedLaneWaypoints(void)const " (?GetAffectedLaneWaypoints@TrafficLight@client@carla@@QBE?AV?$vector@V?$shared_ptr@VWaypoint@client@carla@@@boost@@V?$allocator@V?$shared_ptr@VWaypoint@client@carla@@@boost@@@std@@@std@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::TrafficLight::ResetGroup(void)" (?ResetGroup@TrafficLight@client@carla@@QAEXXZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: class std::vector<class boost::shared_ptr<class carla::client::TrafficLight>,class std::allocator<class boost::shared_ptr<class carla::client::TrafficLight> > > __thiscall carla::client::TrafficLight::GetGroupTrafficLights(void)" (?GetGroupTrafficLights@TrafficLight@client@carla@@QAE?AV?$vector@V?$shared_ptr@VTrafficLight@client@carla@@@boost@@V?$allocator@V?$shared_ptr@VTrafficLight@client@carla@@@boost@@@std@@@std@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: unsigned int __thiscall carla::client::TrafficLight::GetPoleIndex(void)" (?GetPoleIndex@TrafficLight@client@carla@@QAEIXZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: bool __thiscall carla::client::TrafficLight::IsFrozen(void)const " (?IsFrozen@TrafficLight@client@carla@@QBE_NXZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::TrafficLight::Freeze(bool)" (?Freeze@TrafficLight@client@carla@@QAEX_N@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: float __thiscall carla::client::TrafficLight::GetElapsedTime(void)const " (?GetElapsedTime@TrafficLight@client@carla@@QBEMXZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: float __thiscall carla::client::TrafficLight::GetRedTime(void)const " (?GetRedTime@TrafficLight@client@carla@@QBEMXZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::TrafficLight::SetRedTime(float)" (?SetRedTime@TrafficLight@client@carla@@QAEXM@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: float __thiscall carla::client::TrafficLight::GetYellowTime(void)const " (?GetYellowTime@TrafficLight@client@carla@@QBEMXZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::TrafficLight::SetYellowTime(float)" (?SetYellowTime@TrafficLight@client@carla@@QAEXM@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: float __thiscall carla::client::TrafficLight::GetGreenTime(void)const " (?GetGreenTime@TrafficLight@client@carla@@QBEMXZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::TrafficLight::SetGreenTime(float)" (?SetGreenTime@TrafficLight@client@carla@@QAEXM@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: enum carla::rpc::TrafficLightState __thiscall carla::client::TrafficLight::GetState(void)const " (?GetState@TrafficLight@client@carla@@QBE?AW4TrafficLightState@rpc@3@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::TrafficLight::SetState(enum carla::rpc::TrafficLightState)" (?SetState@TrafficLight@client@carla@@QAEXW4TrafficLightState@rpc@3@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > __thiscall carla::client::World::GetNamesOfAllObjects(void)const " (?GetNamesOfAllObjects@World@client@carla@@QBE?AV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::World::ApplyTexturesToObjects(class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > const &,class carla::rpc::Texture<struct carla::sensor::data::Color> const &,class carla::rpc::Texture<struct carla::rpc::FloatColor> const &,class carla::rpc::Texture<struct carla::rpc::FloatColor> const &,class carla::rpc::Texture<struct carla::rpc::FloatColor> const &)" (?ApplyTexturesToObjects@World@client@carla@@QAEXABV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@ABV?$Texture@UColor@data@sensor@carla@@@rpc@3@ABV?$Texture@UFloatColor@rpc@carla@@@73@22@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::World::ApplyTexturesToObject(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,class carla::rpc::Texture<struct carla::sensor::data::Color> const &,class carla::rpc::Texture<struct carla::rpc::FloatColor> const &,class carla::rpc::Texture<struct carla::rpc::FloatColor> const &,class carla::rpc::Texture<struct carla::rpc::FloatColor> const &)" (?ApplyTexturesToObject@World@client@carla@@QAEXABV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@ABV?$Texture@UColor@data@sensor@carla@@@rpc@3@ABV?$Texture@UFloatColor@rpc@carla@@@73@22@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::World::ApplyFloatColorTextureToObjects(class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > > const &,enum carla::rpc::MaterialParameter const &,class carla::rpc::Texture<struct carla::rpc::FloatColor> const &)" (?ApplyFloatColorTextureToObjects@World@client@carla@@QAEXABV?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@std@@ABW4MaterialParameter@rpc@3@ABV?$Texture@UFloatColor@rpc@carla@@@73@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall carla::client::World::ApplyFloatColorTextureToObject(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > const &,enum carla::rpc::MaterialParameter const &,class carla::rpc::Texture<struct carla::rpc::FloatColor> const &)" (?ApplyFloatColorTextureToObject@World@client@carla@@QAEXABV?$basi [... cut ...]
libcarla.obj : error LNK2001: unresolved external symbol __imp__PyEval_RestoreThread
libcarla.obj : error LNK2001: unresolved external symbol __imp__PyUnicode_FromStringAndSize
build\lib.win32-3.9\carla\libcarla.cp39-win_amd64.pyd : fatal error LNK1120: 453 unresolved externals
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x86\\link.exe' failed with exit code 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\carla\PythonAPI\carla\dist"!

PC12138XD, 2024-04-13:
> please, wrap error logs / code into code blocks
> 
> `like this`
> 
> ```
> or like this
> for a multi-line code block
> ```
> 
> to avoid accidentally mentioning users with short usernames such as myself

Thanks, but I don't quite understand how such code should be used, e.g. where should this code be placed?

MarcelPiNacy, 2024-05-03:
Hi,

You could share logs as long as this as an attached file.
On the issue itself: how many python versions do you have installed? Do you have anaconda?

Marcel

ronitsharan, 2024-05-31:
I am also getting the same error

ronitsharan, 2024-05-31:
` Creating library build\temp.win32-3.8\Release\source/libcarla\libcarla.cp38-win_amd64.lib and object build\temp.win32-3.8\Release\source/libcarla\libcarla.cp38-win_amd64.exp
libcarla.obj : error LNK2001: unresolved external symbol _png_destroy_write_struct
libcarla.obj : error LNK2001: unresolved external symbol _png_set_bKGD
libcarla.obj : error LNK2001: unresolved external symbol _png_set_cHRM_fixed
libcarla.obj : error LNK2001: unresolved external symbol __imp__PyBool_FromLong
libcarla.obj : error LNK2001: unresolved external symbol "void __cdecl boost::python::throw_error_already_set(void)" (?throw_error_already_set@python@boost@@YAXXZ)
libcarla.obj : error LNK2001: unresolved external symbol "struct boost::python::self_ns::self_t boost::python::self_ns::self" (?self@self_ns@python@boost@@3Uself_t@123@A)
libcarla.obj : error LNK2001: unresolved external symbol "void * __cdecl boost::python::converter::get_lvalue_from_python(struct _object *,struct boost::python::converter::registration const &)" (?get_lvalue_from_python@converter@python@boost@@YAPAXPAU_object@@ABUregistration@123@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "struct boost::python::converter::registration const & __cdecl boost::python::converter::registry::lookup_shared_ptr(struct boost::python::type_info)" (?lookup_shared_ptr@registry@converter@python@boost@@YAABUregistration@234@Utype_info@34@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "struct boost::python::converter::registration const * __cdecl boost::python::converter::registry::query(struct boost::python::type_info)" (?query@registry@converter@python@boost@@YAPBUregistration@234@Utype_info@34@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "void __cdecl boost::python::converter::registry::push_back(void * (__cdecl*)(struct _object *),void (__cdecl*)(struct _object *,struct boost::python::converter::rvalue_from_python_stage1_data *),struct boost::python::type_info,struct _typeobject const * (__cdecl*)(void))" (?push_back@registry@converter@python@boost@@YAXP6APAXPAU_object@@@ZP6AX0PAUrvalue_from_python_stage1_data@234@@ZUtype_info@34@P6APBU_typeobject@@XZ@Z)
libcarla.obj : error LNK2001: unresolved external symbol "void __cdecl boost::python::converter::registry::insert(struct _object * (__cdecl*)(void const *),struct boost::python::type_info,struct _typeobject const * (__cdecl*)(void))" (?insert@registry@converter@python@boost@@YAXP6APAU_object@@PBX@ZUtype_info@34@P6APBU_typeobject@@XZ@Z)
libcarla.obj : error LNK2001: unresolved external symbol "void __cdecl boost::python::converter::registry::insert(void * (__cdecl*)(struct _object *),void (__cdecl*)(struct _object *,struct boost::python::converter::rvalue_from_python_stage1_data *),struct boost::python::type_info,struct _typeobject const * (__cdecl*)(void))" (?insert@registry@converter@python@boost@@YAXP6APAXPAU_object@@@ZP6AX0PAUrvalue_from_python_stage1_data@234@@ZUtype_info@34@P6APBU_typeobject@@XZ@Z)
libcarla.obj : error LNK2001: unresolved external symbol "struct boost::python::converter::registration const & __cdecl boost::python::converter::registry::lookup(struct boost::python::type_info)" (?lookup@registry@converter@python@boost@@YAABUregistration@234@Utype_info@34@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: void __thiscall boost::python::converter::shared_ptr_deleter::operator()(void const *)" (??Rshared_ptr_deleter@converter@python@boost@@QAEXPBX@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: __thiscall boost::python::converter::shared_ptr_deleter::~shared_ptr_deleter(void)" (??1shared_ptr_deleter@converter@python@boost@@QAE@XZ)
libcarla.obj : error LNK2001: unresolved external symbol "public: __thiscall boost::python::converter::shared_ptr_deleter::shared_ptr_deleter(class boost::python::handle<struct _object>)" (??0shared_ptr_deleter@converter@python@boost@@QAE@V?$handle@U_object@@@23@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "void __cdecl boost::python::converter::void_result_from_python(struct _object *)" (?void_result_from_python@converter@python@boost@@YAXPAU_object@@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "void * __cdecl boost::python::converter::rvalue_from_python_stage2(struct _object *,struct boost::python::converter::rvalue_from_python_stage1_data &,struct boost::python::converter::registration const &)" (?rvalue_from_python_stage2@converter@python@boost@@YAPAXPAU_object@@AAUrvalue_from_python_stage1_data@123@ABUregistration@123@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "struct boost::python::converter::rvalue_from_python_stage1_data __cdecl boost::python::converter::rvalue_from_python_stage1(struct _object *,struct boost::python::converter::registration const &)" (?rvalue_from_python_stage1@converter@python@boost@@YA?AUrvalue_from_python_stage1_data@123@PAU_object@@ABUregistration@123@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "public: __thiscall boost::python::converter::detail::arg_to_python_base::arg_to_python_base(void const volatile *,struct boost::python::converter::registration const &)" (??0arg_to_python_base@detail@converter@python@boost@@QAE@PDXABUregistration@234@@Z)
libcarla.obj : error LNK2001: unresolved external symbol "struct _object * __cdecl boost::python::converter::do_return_to_python(char const *)" (?do_return_to_python@converter@python@boost@@YAPAU_object@@PBD@Z)
[... cut ...]
libcarla.obj : error LNK2001: unresolved external symbol __imp__PyEval_RestoreThread
libcarla.obj : error LNK2001: unresolved external symbol __imp__PyUnicode_FromStringAndSize
build\lib.win32-3.8\carla\libcarla.cp38-win_amd64.pyd : fatal error LNK1120: 468 unresolved externals
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x86\\link.exe' failed with exit status 1120

-[BuildPythonAPI]: Carla lib for python has been successfully installed in "C:\Users\ronit\carla\PythonAPI\carla\dist"!
`

lnavneet331, 2024-11-29:
Is anyone able to fix this issue? I am getting the same error.

YeZibn, 2024-12-25:
I changed the python version from 32-bit to 64-bit and it worked