carla-simulator/carla #7188: How to visualize multiple Lidar data at the same time

eyabesbes, 2024-02-27:
Hi everyone!
I have used and modified the open3d-lidar.py file existing within the pythonAPI folder in Carla to visualize the Lidar data and the BEV map (which is a projection of the Lidar data onto a 2D plane) of a certain ego-vehicle.
I have also created some 'static obstacles' and "dynamic obstacles".
Though, now I'm having trouble to visualize the Lidar data of every dynamic vehicle at the same time with the lidar data of the ego_vehicle. I Have tried to attach to every vehicle a lidar sensor. But then I did not know what to do. It would be absurd if I define the callback-function for every sensor (which is basically the same code). I have tried to retrieve point clouds of every vehicle and aggregate them but I don't how to do that.
```
#!/usr/bin/env python

# Copyright (c) 2020 Computer Vision Center (CVC) at the Universitat Autonoma de
# Barcelona (UAB).
#
# This work is licensed under the terms of the MIT license.
# For a copy, see <https://opensource.org/licenses/MIT>.

"""Open3D Lidar visuialization example for CARLA"""

import glob
import os
import sys
import argparse
import time
from datetime import datetime
import random
import numpy as np
from matplotlib import cm
import open3d as o3d
import random

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla

VIRIDIS = np.array(cm.get_cmap('plasma').colors)
VID_RANGE = np.linspace(0.0, 1.0, VIRIDIS.shape[0])
LABEL_COLORS = np.array([
    (255, 255, 255), # None
    (70, 70, 70),    # Building
    (100, 40, 40),   # Fences
    (55, 90, 80),    # Other
    (220, 20, 60),   # Pedestrian
    (153, 153, 153), # Pole
    (157, 234, 50),  # RoadLines
    (128, 64, 128),  # Road
    (244, 35, 232),  # Sidewalk
    (107, 142, 35),  # Vegetation
    (0, 0, 142),     # Vehicle
    (102, 102, 156), # Wall
    (220, 220, 0),   # TrafficSign
    (70, 130, 180),  # Sky
    (81, 0, 81),     # Ground
    (150, 100, 100), # Bridge
    (230, 150, 140), # RailTrack
    (180, 165, 180), # GuardRail
    (250, 170, 30),  # TrafficLight
    (110, 190, 160), # Static
    (170, 120, 50),  # Dynamic
    (45, 60, 150),   # Water
    (145, 170, 100), # Terrain
]) / 255.0 # normalize each channel [0-1] since is what Open3D uses

def lidar_callback(point_cloud, lidar_point_list, bev_point_list):
    """Prepares a point cloud with intensity
    colors ready to be consumed by Open3D"""
    data = np.copy(np.frombuffer(point_cloud.raw_data, dtype=np.dtype('f4')))
    data = np.reshape(data, (int(data.shape[0] / 4), 4))
[... 303 lines cut ...]
        default=1000000,
        type=int,
        help='lidar\'s points per second (default: 500000)')
    argparser.add_argument(
        '-x',
        default=0.0,
        type=float,
        help='offset in the sensor position in the X-axis in meters (default: 0.0)')
    argparser.add_argument(
        '-y',
        default=0.0,
        type=float,
        help='offset in the sensor position in the Y-axis in meters (default: 0.0)')
    argparser.add_argument(
        '-z',
        default=0.0,
        type=float,
        help='offset in the sensor position in the Z-axis in meters (default: 0.0)')
    args = argparser.parse_args()

    try:
        main(args)
    except KeyboardInterrupt:
        print(' - Exited by user.')
```

csonthomisi, 2024-02-28:
I think the first problem here is that you may use different "role_name" for each vehicle and also for each sensor.
My code is the following, which works with multiple lidars at the same time:
```
        bp_lid_1 = blueprint_library.find("sensor.lidar.ray_cast")
        lidar_1_settings = {
            "role_name": "sync/Lidar_1",
            "channels": "256",
            "range": "150",
            "points_per_second": str(256 * 20 * 60),
            "rotation_frequency": "20",
            "upper_fov": "0.0",
            "lower_fov": "-45.0",
            "horizontal_fov": "60",
        }
        for key, value in lidar_1_settings.items():
            bp_lid_1.set_attribute(str(key), str(value))
        lidar_1 = world.spawn_actor(
            bp_lid_1,
            sensor_station_1_lidar,
            attach_to=None,
        )
```

Also i think you should use the "attach_to" property, to place the sensors for each vehicle.

eyabesbes, 2024-02-28:
I have understood from your explanation that assigning a role to each Lidar sensor allows for retrieving multiple Lidar data streams from sensors that share the same role right ? Therefore, I have made slight modifications to the code:

- add this line in the generate lidar function:

```lidar_bp.set_attribute("role_name", str("sync/Lidar_1")) ```
so here's the modified function
```
def generate_lidar_bp(arg, world, blueprint_library, delta):
    """Generates a CARLA blueprint based on the script parameters"""
   
    lidar_bp = world.get_blueprint_library().find('sensor.lidar.ray_cast')
    lidar_bp.set_attribute("role_name", str("sync/Lidar_1")) 
    
    lidar_bp.set_attribute('noise_stddev', '0.2')

    lidar_bp.set_attribute('upper_fov', str(90))
    lidar_bp.set_attribute('lower_fov', str(-90))
    lidar_bp.set_attribute('channels', str(arg.channels))
    lidar_bp.set_attribute('range', str(500))
    lidar_bp.set_attribute('rotation_frequency', str(1.0 / delta))
    lidar_bp.set_attribute('points_per_second', str((1000000)))
    return lidar_bp
```

- add a loop to generate multiple vehicles and attach the lidar to each vehicle:
```
blueprint_library = world.get_blueprint_library()
        lidar_bp = generate_lidar_bp(arg, world, blueprint_library, delta)
        user_offset = carla.Location(arg.x, arg.y, arg.z)
        lidar_transform = carla.Transform(carla.Location(x=-0.5, z=1.8) + user_offset)
        blueprints = blueprint_library.filter('vehicle.*')

        spawn_points = world.get_map().get_spawn_points()
        
            
        for i in range(100):
            spawn_point = spawn_points[i]
            blueprint = random.choice(blueprints)
            vehicle1 = world.spawn_actor(blueprint, spawn_point)
            vehicle1.set_simulate_physics(False)
            vehicle1.set_autopilot(True)
            lidar = world.spawn_actor(lidar_bp, lidar_transform, attach_to=vehicle1)
        lidar.listen(lambda data: lidar_callback(data, lidar_point_list, bev_point_list))
```
the problem here is that when executing the code:
Unfortunately, I only get the lidar data of one vehicle:
and this is the warnings that I keep getting:
```WARNING:sensor object went out of the scope but the sensor is still alive in the simulation: Actor 557 (sensor.lidar.ray_cast) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 559 (sensor.lidar.ray_cast)```
![image](https://github.com/carla-simulator/carla/assets/95165444/7517c62e-bcd6-4fc1-a742-ecc6c2117c93)

eyabesbes, 2024-02-28:
> I think the first problem here is that you may use different "role_name" for each vehicle and also for each sensor. My code is the following, which works with multiple lidars at the same time: [...]

I have understood from your explanation that assigning a role to each Lidar sensor allows us to retrieve multiple Lidar data streams from sensors that share the same role right ? Therefore, I have made slight modifications to the code:

- add this line in the generate lidar function:

```lidar_bp.set_attribute("role_name", str("sync/Lidar_1")) ```
so here's the modified function
```
def generate_lidar_bp(arg, world, blueprint_library, delta):
    """Generates a CARLA blueprint based on the script parameters"""
   
    lidar_bp = world.get_blueprint_library().find('sensor.lidar.ray_cast')
    lidar_bp.set_attribute("role_name", str("sync/Lidar_1")) 
    
    lidar_bp.set_attribute('noise_stddev', '0.2')

    lidar_bp.set_attribute('upper_fov', str(90))
    lidar_bp.set_attribute('lower_fov', str(-90))
    lidar_bp.set_attribute('channels', str(arg.channels))
    lidar_bp.set_attribute('range', str(500))
    lidar_bp.set_attribute('rotation_frequency', str(1.0 / delta))
    lidar_bp.set_attribute('points_per_second', str((1000000)))
    return lidar_bp
```

- add a loop to generate multiple vehicles and attach the lidar to each vehicle:
```
blueprint_library = world.get_blueprint_library()
        lidar_bp = generate_lidar_bp(arg, world, blueprint_library, delta)
        user_offset = carla.Location(arg.x, arg.y, arg.z)
        lidar_transform = carla.Transform(carla.Location(x=-0.5, z=1.8) + user_offset)
        blueprints = blueprint_library.filter('vehicle.*')

        spawn_points = world.get_map().get_spawn_points()
        
            
        for i in range(100):
            spawn_point = spawn_points[i]
            blueprint = random.choice(blueprints)
            vehicle1 = world.spawn_actor(blueprint, spawn_point)
            vehicle1.set_simulate_physics(False)
            vehicle1.set_autopilot(True)
            lidar = world.spawn_actor(lidar_bp, lidar_transform, attach_to=vehicle1)
        lidar.listen(lambda data: lidar_callback(data, lidar_point_list, bev_point_list))
```
the problem here is that when executing the code:
Unfortunately, I only get the lidar data of one vehicle:
and this is the warnings that I keep getting:
```WARNING:sensor object went out of the scope but the sensor is still alive in the simulation: Actor 557 (sensor.lidar.ray_cast) 
WARNING: sensor object went out of the scope but the sensor is still alive in the simulation: Actor 559 (sensor.lidar.ray_cast)```
![image](https://github.com/carla-simulator/carla/assets/95165444/7517c62e-bcd6-4fc1-a742-ecc6c2117c93)

eyabesbes, 2024-02-28:
> I think the first problem here is that you may use different "role_name" for each vehicle and also for each sensor. My code is the following, which works with multiple lidars at the same time: [...]

here's the full modified code
```
#!/usr/bin/env python

# Copyright (c) 2020 Computer Vision Center (CVC) at the Universitat Autonoma de
# Barcelona (UAB).
#
# This work is licensed under the terms of the MIT license.
# For a copy, see <https://opensource.org/licenses/MIT>.

"""Open3D Lidar visuialization example for CARLA"""

import glob
import os
import sys
import argparse
import time
from datetime import datetime
import random
import numpy as np
from matplotlib import cm
import open3d as o3d
import random

try:
    sys.path.append(glob.glob('../carla/dist/carla-*%d.%d-%s.egg' % (
        sys.version_info.major,
        sys.version_info.minor,
        'win-amd64' if os.name == 'nt' else 'linux-x86_64'))[0])
except IndexError:
    pass

import carla

VIRIDIS = np.array(cm.get_cmap('plasma').colors)
VID_RANGE = np.linspace(0.0, 1.0, VIRIDIS.shape[0])
LABEL_COLORS = np.array([
    (255, 255, 255), # None
    (70, 70, 70),    # Building
    (100, 40, 40),   # Fences
    (55, 90, 80),    # Other
    (220, 20, 60),   # Pedestrian
    (153, 153, 153), # Pole
    (157, 234, 50),  # RoadLines
    (128, 64, 128),  # Road
    (244, 35, 232),  # Sidewalk
    (107, 142, 35),  # Vegetation
    (0, 0, 142),     # Vehicle
    (102, 102, 156), # Wall
    (220, 220, 0),   # TrafficSign
    (70, 130, 180),  # Sky
    (81, 0, 81),     # Ground
    (150, 100, 100), # Bridge
    (230, 150, 140), # RailTrack
    (180, 165, 180), # GuardRail
    (250, 170, 30),  # TrafficLight
    (110, 190, 160), # Static
    (170, 120, 50),  # Dynamic
    (45, 60, 150),   # Water
    (145, 170, 100), # Terrain
]) / 255.0 # normalize each channel [0-1] since is what Open3D uses

# Your code with corrections

# (Code unchanged above)

def lidar_callback(point_cloud, lidar_point_list, bev_point_list):
    """Prepares a point cloud with intensity
[... 273 lines cut ...]
        default=1000000,
        type=int,
        help='lidar\'s points per second (default: 500000)')
    argparser.add_argument(
        '-x',
        default=0.0,
        type=float,
        help='offset in the sensor position in the X-axis in meters (default: 0.0)')
    argparser.add_argument(
        '-y',
        default=0.0,
        type=float,
        help='offset in the sensor position in the Y-axis in meters (default: 0.0)')
    argparser.add_argument(
        '-z',
        default=0.0,
        type=float,
        help='offset in the sensor position in the Z-axis in meters (default: 0.0)')
    args = argparser.parse_args()

    try:
        main(args)
    except KeyboardInterrupt:
        print(' - Exited by user.')
```

csonthomisi, 2024-02-29:
I think you would have to call the lidar's callback for each lidar in a loop after generation.

eyabesbes, 2024-02-29:
> I think you would have to call the lidar's callback for each lidar in a loop after generation.

Thanks again for your reply. I've spawned multiple vehicles, but I've only attached a LiDAR sensor to two of them. I've created a list to store all the LiDAR sensors I've used and then called the LiDAR callback for each one.

Now, I'm encountering an issue: I can't visualize both LiDAR data sets simultaneously.

https://github.com/carla-simulator/carla/assets/95165444/4c4942e3-c34e-43af-8246-b300ee00e072

I've tried to aggregate the two point clouds at each tick and visualize them in the same window using Open3D. I attempted to define a callback function for each LiDAR and add the point cloud to the o3d.visualization.Visualizer(), but it didn't work as expected. Additionally, I added a return statement to the callback function to return the LiDAR point cloud of a specific vehicle so that I could aggregate it with the point clouds of the second vehicle, but that also didn't work.

Do you have any ideas on how I can solve this issue?

csonthomihaly, 2024-02-29:
Unfortunately I’m not familiar with o3d. 
For me at first look your “while True:” loop looks strange.. 
Also with the callback function i think you just always overwrites the data. 
`i.listen(lambda data: lidar_callback(data, lidar_point_list, bev_point_list)) `
Maybe use a list to store callback data and use later that

csonthomisi, 2024-02-29:
Unfortunately I’m not familiar with o3d. 
For me at first look your “while True:” loop looks strange.. 
Also with the callback function i think you just always overwrites the data. 
`i.listen(lambda data: lidar_callback(data, lidar_point_list, bev_point_list)) `
Maybe use a list to store callback data and use later that

MarcelPiNacy, 2024-05-03:
Hi,

Sorry for the late reply, do you still have this problem?

Marcel