carla-simulator/carla #7192: Problem with Update.sh

kesun0111, 2024-02-29:
I tried to install 0.9.15 release and when I run update.sh, there seems to be some s3 related issues. Here is the terminal output: sudo ./Update.sh 
Backing up existing Content...
renamed '/home/ubuntu/carla/Unreal/CarlaUE4/Content/Carla' -> '/home/ubuntu/carla/Unreal/CarlaUE4/Content/Carla_20240229074713'
--2024-02-29 07:47:13--  http://carla-assets.s3.amazonaws.com/20231108_c5101a5.tar.gz
Resolving carla-assets.s3.amazonaws.com (carla-assets.s3.amazonaws.com)... 52.217.129.49, 52.217.199.121, 54.231.233.9, ...
Connecting to carla-assets.s3.amazonaws.com (carla-assets.s3.amazonaws.com)|52.217.129.49|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2024-02-29 07:47:13 ERROR 404: Not Found.
Has anyone met this issue before?

Paulenoch, 2024-02-29:
I have the same issue, too. It didn't happen when i installed carla last month. I guess it's a Internet problem.

Kirin2003, 2024-02-29:
Same issue. It looks like the file doesn't exist.
[image]

SVKBlackDeath, 2024-02-29:
I have the same issue too, none of the AWS links provided in the ContentVersions.txt work.

mesakas, 2024-03-02:
Same issue.

I use `Udpate.sh`:

```shell
03/02 17:13:28 [ERROR] CUID#7 - Download aborted. URI=http://carla-assets.s3.amazonaws.com/20231108_c5101a5.tar.gz
Exception: [AbstractCommand.cc:351] errorCode=3 URI=http://carla-assets.s3.amazonaws.com/20231108_c5101a5.tar.gz
  -> [HttpSkipResponseCommand.cc:218] errorCode=3 Resource not found

03/02 17:13:28 [NOTICE] Download GID#2b16b3402c00930a not complete: /home/ming/Desktop/carla/Content.tar.gz

Download Results:
gid   |stat|avg speed  |path/URI
======+====+===========+=======================================================
2b16b3|ERR |       0B/s|/home/ming/Desktop/carla/Content.tar.gz

Status Legend:
(ERR):error occurred.

aria2 will resume download if the transfer is restarted.
If there are any errors, then see the log file. See '-l' option in help/man page for details.

```

and I use `https://carla-assets.s3.amazonaws.com/20231108_c5101a5.tar.gz` download the assets:
```xml
<Error>
<Code>NoSuchBucket</Code>
<Message>The specified bucket does not exist</Message>
<BucketName>carla-assets</BucketName>
<RequestId>C92BHYKH6BRYNA0N</RequestId>
<HostId>ZGcLa6SfvNoBwMxaMRbCXmpaSKCoXUDiQOY8S/z0+YMsld6zW7viD4zQ6BTqBV60ZfXbmBCi6OI=</HostId>
</Error>
```

Who can fix this issue?

mesakas, 2024-03-02:
And I tried URL of all version in `~/carla/Util/ContentVersions.txt`:

https://carla-assets.s3.amazonaws.com/20190404_c7b464a.tar.gz
https://carla-assets.s3.amazonaws.com/20190710_0097e66.tar.gz
https://carla-assets.s3.amazonaws.com/20191221_c88604b.tar.gz
https://carla-assets.s3.amazonaws.com/20200306_06b6cb1.tar.gz
https://carla-assets.s3.amazonaws.com/20200422_ea5179a.tar.gz
https://carla-assets.s3.amazonaws.com/20200925_88f9ceb.tar.gz
https://carla-assets.s3.amazonaws.com/20201222_232b876.tar.gz
https://carla-assets.s3.amazonaws.com/20210730_564bcdc.tar.gz
https://carla-assets.s3.amazonaws.com/20211112_d5cfa12.tar.gz
https://carla-assets.s3.amazonaws.com/20221201_5ec9328.tar.gz
https://carla-assets.s3.amazonaws.com/20231108_c5101a5.tar.gz

> In ~/carla/Util/ContentVersions.txt:
>0.9.5: 20190404_c7b464a
>0.9.6: 20190710_0097e66
>0.9.7: 20191221_c88604b
>0.9.8: 20200306_06b6cb1
>0.9.9: 20200422_ea5179a
>0.9.10: 20200925_88f9ceb
>0.9.11: 20201222_232b876
>0.9.12: 20210730_564bcdc
>0.9.13: 20211112_d5cfa12
>0.9.14: 20221201_5ec9328
>0.9.15: 20231108_c5101a5
>Latest: 20231108_c5101a5

None of it works...

wucong-Yy, 2024-03-02:
You can directly obtain development assets in the carla root directory, then skip update.sh and make launch `directly`
`git clone https://bitbucket.org/carla-simulator/carla-content Unreal/CarlaUE4/Content/Carla`

germanros1987, 2024-03-02:
Hi,

Please noticed that the urls of the contents have changed. The latest urls
are in the dev branch.

You can use dev or you can clone the bitbucket repo as an alternative.

Sorry for the inconvenience.

HyzenthlayER, 2024-03-04:
I tried to build 0.9.13 release and I cloned the content files using `git clone https://bitbucket.org/carla-simulator/carla-content Unreal/CarlaUE4/Content/Carla`
However, during the `make launch` process, lots of errors occurred such as `LogAssetRegistry: Error: Package ../../../../CARLA_0.9.13/Unreal/CarlaUE4/Content/Carla/Maps/Town12/LaneMarking2_Spec.uasset has malformed tag`. It seems that some of the files are broken. Does anyone know how to fix it?

Kirin2003, 2024-03-08:
> Same issue. It looks like the file doesn't exist. ![Snipaste_2024-02-29_18-28-09](https://private-user-images.githubusercontent.com/89304278/308888581-70ce2dc4-c2b9-4fad-a25c-75a81d65da40.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk5MDIwNTUsIm5iZiI6MTcwOTkwMTc1NSwicGF0aCI6Ii84OTMwNDI3OC8zMDg4ODg1ODEtNzBjZTJkYzQtYzJiOS00ZmFkLWEyNWMtNzVhODFkNjVkYTQwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA4VDEyNDIzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEwZjljMjNjNjBlMjM0ZDAyOTNhOWIxNTdmZjJhZDE3ZWY0YzUyNGZhNWJhNGQ4Y2Q3ZWNjODA4NWVhOWJjOGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.dEBxmDSQGtTeAKKIeGFrm4k8GC1hWgHSsZJkcwh3ERc)

According to [this response](https://github.com/carla-simulator/carla/issues/7196#issuecomment-1976496048), the solution is to change the 50th line of `Update.sh`: `https://carla-assets.s3.us-east-005.backblazeb2.com/${CONTENT_ID}.tar.gz`. That works for me.

closed 2024-03-14 by Blyron

mohussein97, 2024-05-06:
> Hi, Please noticed that the urls of the contents have changed. The latest urls are in the dev branch. You can use dev or you can clone the bitbucket repo as an alternative. Sorry for the inconvenience.
> […](#)

the dev branch contains .9.15 I want .9.14 and .9.13

JDIRK3, 2024-05-22:
> I tried to build 0.9.13 release and I cloned the content files using `git clone https://bitbucket.org/carla-simulator/carla-content Unreal/CarlaUE4/Content/Carla` However, during the `make launch` [...]

Did you solve this problem?

ChichengZZZ, 2024-05-30:
> > Hi, Please noticed that the urls of the contents have changed. The latest urls are in the dev branch. You can use dev or you can clone the bitbucket repo as an alternative. Sorry for the inconvenience.
> > […](#)
> 
> the dev branch contains .9.15 I want .9.14 and .9.13

me too, I want calra-content 0.9.13

muety, 2024-06-19:
Same problem here (for `http://carla-assets.s3.amazonaws.com/20231108_c5101a5.tar.gz`), could somebody have a look at fixing this, please?

Blyron, 2024-06-19:
CARLA S3 amazon web services IS NOT AVAILABLE ANYMORE. You need to change manually new s3 backblaze services

https://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz

MarceloAdas, 2024-07-10:
Same question: I tried to build 0.9.13 release and I cloned the content files using git clone https://bitbucket.org/carla-simulator/carla-content Unreal/CarlaUE4/Content/Carla
However, during the make launch process, lots of errors occurred such as LogAssetRegistry: Error: Package ../../../../CARLA_0.9.13/Unreal/CarlaUE4/Content/Carla/Maps/Town12/LaneMarking2_Spec.uasset has malformed tag. It seems that some of the files are broken. Has anyone solved the problem

Blyron, 2024-07-10:
Have you installed Git LFS?

MarceloAdas, 2024-07-10:
> CARLA S3 amazon web services IS NOT AVAILABLE ANYMORE. You need to change manually new s3 backblaze services
> 
> https://carla-assets.s3.us-east-005.backblazeb2.com/20231108_c5101a5.tar.gz

I have solved the problem through this method, thank you

Haobo-Liu, 2024-07-20:
https://carla-releases.s3.us-east-005.backblazeb2.com/Linux/CARLA_0.9.13.tar.gz
try it,change 0.9.13 to your own version

Haobo-Liu, 2024-07-20:
> And I tried URL of all version in `~/carla/Util/ContentVersions.txt`:
>
> https://carla-assets.s3.amazonaws.com/20190404_c7b464a.tar.gz https://carla-assets.s3.amazonaws.com/20190710_0097e66.tar.gz [...]

https://carla-releases.s3.us-east-005.backblazeb2.com/Linux/CARLA_0.9.13.tar.gz
try it,change 0.9.13 to your own version

MasoudJTehrani, 2025-01-24:
The `./update.sh` problem is solved in [Carla 9.15.2](https://github.com/carla-simulator/carla/tree/0.9.15.2)

`git clone --depth 1 -b 0.9.15.2 https://github.com/carla-simulator/carla.git`

SDlearner-maker, 2025-09-29:
> Hi,
> 
> Please noticed that the urls of the contents have changed. The latest urls
> are in the dev branch.
> 
> You can use dev or you can clone the bitbucket repo as an alternative.
> 
> Sorry for the inconvenience.
> […](#)

Hello,

THank you for the reply. However, in the [https://github.com/carla-simulator/carla/tree/ue4-dev](url) branch the URL is not mentioned, I am trying to build for CARLA 0.9.15. As the downlaoding of assets is necessary, therefore I will need to make sure if the URLs mentioned are compatible as well.

Thanks!

qujue720, 2025-11-06:
> > 同样的问题。该文件似乎不存在。![Snipaste_2024-02-29_18-28-09](https://private-user-images.githubusercontent.com/89304278/308888581-70ce2dc4-c2b9-4fad-a25c-75a81d65da40.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk5MDIwNTUsIm5iZiI6MTcwOTkwMTc1NSwicGF0aCI6Ii84OTMwNDI3OC8zMDg4ODg1ODEtNzBjZTJkYzQtYzJiOS00ZmFkLWEyNWMtNzVhODFkNjVkYTQwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA4VDEyNDIzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWEwZjljMjNjNjBlMjM0ZDAyOTNhOWIxNTdmZjJhZDE3ZWY0YzUyNGZhNWJhNGQ4Y2Q3ZWNjODA4NWVhOWJjOGMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.dEBxmDSQGtTeAKKIeGFrm4k8GC1hWgHSsZJkcwh3ERc)
> 
> 根据[这个回应](https://github.com/carla-simulator/carla/issues/7196#issuecomment-1976496048)，解决方案是更改 ： 的第 50 行。这对我有用。`Update.sh``https://carla-assets.s3.us-east-005.backblazeb2.com/${CONTENT_ID}.tar.gz`

你的回答对我很有用，感谢。