carla-simulator/carla #7197: Obtaining ground truth occupancy grids

carlostojal, 2024-02-29:
Hi.
I am developing a project that needs the ground truth occupancy grid of the map (either 2D or 3D, preferably the latter), but I couldn't find any documentation regarding that. Does anyone know how can this be achieved? Thanks in advance.

yhh-IV, 2024-03-04:
Currently I have used Lidar sensor to obtain the occupancy grid. But I still expect for a better choice.

PatrickPromitzer, 2024-03-07:
It depends on the data you need.
With the Semantic LIDAR, you get the semantic tag (house, ground, road, ...) and index of the object.
[https://carla.readthedocs.io/en/latest/ref_sensors/#semantic-lidar-sensor](https://carla.readthedocs.io/en/latest/ref_sensors/#semantic-lidar-sensor)
 
For the camera, there exists a semantic version, too.
[https://carla.readthedocs.io/en/latest/ref_sensors/#semantic-segmentation-camera](https://carla.readthedocs.io/en/latest/ref_sensors/#semantic-segmentation-camera)

What kind of data do you need for your occupancy grid?

Sreesruth, 2024-03-21:
hello, what is the issue that is to be fixed ??

MarcelPiNacy, 2024-05-03:
Hi,

Should I close this issue? Or do you still need help with this?

Marcel

Sreesruth, 2024-05-03:
please close it, thank you
will ir be alright to ask any questions after closing the issue ??

CallMeFrozenBanana, 2024-10-15:
> Currently I have used Lidar sensor to obtain the occupancy grid. But I still expect for a better choice.

Hi, Have you met any point-clouds aligning problems when you merged point-clouds of different frame?