carla-simulator/carla #7201: Carla code working for in-build map but not in RoadRunner map

SExpert12, 2024-03-01:
I have created map with RoadRunner map but I can run the script of manual_control and automatic_control perfectly in created RoadRunner map. But this is the code which runs perfectly with carla in-build map but not in the map which is created with roadrunner. How to resolved this issue?

import carla
import random

client= carla.Client('localhost',2000)
client.set_timeout(0.1)
world= client.get_world()

settings = world.get_settings()
settings.synchronous_mode = True
settings.fixed_delta_seconds = 0.05
world.apply_settings(settings)

tm= client.get_trafficmanager()
tm.set_synchronous_mode(True)

tm.set_random_device_seed(0)
random.seed(0)

spectator = world.get_spectator()

sps = world.get_map().get_spawn_points()

for i, sp in enumerate(sps):
	world.debug.draw_string(sp.location, str(i),life_time = 10)
models = ['dodge', 'audi', 'model3', 'mini', 'mustang', 'lincoln', 'prius', 'nissan', 'crown', 'impala']
blueprints = []
for vehicle in world.get_blueprint_library().filter('*vehicle*'):
    if any(model in vehicle.id for model in models):
        blueprints.append(vehicle)

maxveh=50
maxveh=min([maxveh,len(sps)])
vehicles= []

for i, sp in enumerate(random.sample(sps,maxveh)):
	temp= world.try_spawn_actor(random.choice(blueprints),sp)
	if temp is not None:
		vehicles.append(temp)

for vehicle in vehicles:
	vehicle.set_autopilot(True)
	tm.distance_to_leading_vehicle(vehicle, 5.0)
	tm.ignore_lights_percentage(vehicle,random.randint(0,50))

while True:
	world.tick(5)

csonthomisi, 2024-03-06:
If I understand you correctly, you have created a map in RoadRunner, which when loaded does not run the manual_control.py and automatic_control.py scripts. Do I get it right?

I experienced the same thing, I tried several things but I couldn't find out clearly what is causing the error.
- My first question is, is the map created in RoadRunner, is it at 0 meters in altitude value, or is it at a different altitude?

I got this error when I imported a map that was at a real (170m) altitude.
The working solution for me was to export the RoadRunner .fbx file at the real (170m) elevation and then project the roads in RoadRunner down to elevation 0m. From this I exported the .xodr file (choosing the Unreal version rather than CARLA when exporting).

So after importing, the waypoints and spawnpoints on the map are also set to 0 meters in height, which I placed above the road surface in the Unreal Editor.

I suspect that the waypoints cannot be generated correctly above 0 meters height, and that is why the two files crash out. If you skip the part in the script that does the control the script will start but you can't control the car.
At least that's how it worked for me.

SExpert12, 2024-04-30:
No manual_control.py and automatic_control.py run perfectly with my generated roadRunner map. Only this code is not working for the created map. This code runs perfectly for CARLA in build map.