carla-simulator/carla #7202: behavior of set_target_velocity

XiaoLiSean, 2024-03-04:
Hi, everyone. I recently encountered a problem using set_target_velocity(). I'm using Carla 0.9.15 in Ubuntu 20.04.6. I know it might be a little bit uncommon, however, my development requires a pipeline as follows: set the vehicle speed to zero -> complete a certain task -> accelerate the vehicle speed to a desired value. Please see the code attached. 

```
import argparse, sys
import carla

if __name__ == '__main__':

    # Get the host and port id
    argparser = argparse.ArgumentParser(description=__doc__)
    argparser.add_argument('--host', metavar='H', default='127.0.0.1',
        help='IP of the host server (default: 127.0.0.1)')
    argparser.add_argument('-p', '--port', metavar='P', default=2000, type=int,
        help='TCP port to listen to (default: 2000)')
    argparser.add_argument('--tm-port', metavar='P', default=8000, type=int,
        help='Port to communicate with TM (default: 8000)')
    argparser.add_argument('--record', action='store_true', help='Record video')
    args = argparser.parse_args()

    try:
        dt = 0.01
        # Connect to the client
        client = carla.Client(args.host, args.port)
        client.set_timeout(10.0)
        client.load_world('Town06')

        # Retrieve the world object
        world = client.get_world()
        settings = world.get_settings()
        settings.synchronous_mode = True
        settings.fixed_delta_seconds = dt
        world.apply_settings(settings)

        # Retrieve the spectator object and change the view point and angle
        spectator = world.get_spectator()
        transform = spectator.get_transform()
        transform.location = carla.Location(x=250, y=260, z=25.50) # left half of the view
        transform.rotation = carla.Rotation(roll=0, pitch=-60, yaw=-90)
        spectator.set_transform(transform)

        # Intialize the traffic manager
        traffic_manager = client.get_trafficmanager(args.tm_port)
        traffic_manager.set_synchronous_mode(True)
        world.tick() # this is neccesary to update the actor info
        
        # Initialize ego vehicle
        blueprint = world.get_blueprint_library().find('vehicle.lincoln.mkz_2020')
        blueprint.set_attribute('role_name', 'hero')
        transform = carla.Transform()
        transform.rotation = carla.Rotation(roll=0, pitch=0, yaw=0)
        transform.location = carla.Location(x=250, y=244.55, z=0.024)
        car = world.spawn_actor(blueprint, transform)
        world.tick() # this is neccesary to update the actor info

        # Accelerate to set speed
        v_k = 40 # in [m/s]
        a_k = 1 # in [m/s^2]
        car.set_target_velocity(carla.Vector3D(x=v_k)) # set_target_velocity in [m/s]
        car.enable_constant_velocity(carla.Vector3D(x=v_k)) # enforce set speed
        while True:
            car.set_target_velocity(carla.Vector3D(x=0.0)) # set_target_velocity in [m/s]
            car.enable_constant_velocity(carla.Vector3D(x=0.0)) # enforce set speed
            world.tick()

            # ---------------------------------------
            # some other code blocks to be execute after setting vehicle speed to zero
            # ---------------------------------------
            # ......................................
            # ---------------------------------------

            # accelerate
            v_k += a_k*dt
            car.set_target_velocity(carla.Vector3D(x=v_k)) # set_target_velocity in [m/s]
            car.enable_constant_velocity(carla.Vector3D(x=v_k)) # enforce set speed

            # settings = world.get_settings()
            # settings.fixed_delta_seconds = dt
            # world.apply_settings(settings)
            world.tick()
            # acceleration not working at larger speed
            print('Desired / Actual: {}/{}'.format(v_k, car.get_velocity().x))
            v_k = car.get_velocity().x

    except KeyboardInterrupt:
        print('Cancelled by user...')

    except Exception as err:
        print('Error: {}'.format(err))

    # Terminate carla and disconnet the port
    finally:
        pass
```
However, this piece of code cannot set the velocity (which remains zero instead) to the desired ones with `world.tick()` after a few loops. When I tried to uncomment the following lines, I somehow could get non-zero velocity, but the velocity is decreasing while I anticipate it to increase.
```         
# settings = world.get_settings()
# settings.fixed_delta_seconds = dt
# world.apply_settings(settings)
```
I'm personally not very familiar with the underlying Carla development, and explanations of these observations from experts will be much appreciated. Meanwhile,  I would be very grateful if I could possibly be provided with solutions to these issues or guidance for realizing the aforementioned pipeline (i.e., set the vehicle speed to zero -> complete a certain task -> accelerate the vehicle speed to a desired value).

Thank you very much!

PatrickPromitzer, 2024-03-26:
I didn't try the code for now (maybe I do it later), but I have a guess.

In Carla, Vehicles have a speed limit, for example, 50km/h.
I don't know the limit of each car, but my guess is, that if you set 40m/s, it could be over the speed limit of the car.

One other way to get to a higher speed would be using the add_impulse() function.

```python
speed = 30 # I didn't test if it is m/s or km/h
# The friction will push the vehicle in a random direction if the vehicle is on the floor
actor_transform: carla.Transform = car.get_transform()
actor_transform.rotation.roll = 0.0
actor_transform.rotation.pitch = 0.0
actor_transform.location.z += 0.1
car.set_transform(actor_transform)

# global xyz for adding impulse
impulse_strength = 1000.0 * speed
push_factor = []
for yaw in [actor_transform.rotation.yaw + 90, actor_transform.rotation.yaw]:
    while yaw > 180 or yaw < -180:
        if yaw > 180:
            yaw -= 360
        elif yaw < -180:
            yaw += 360

    if yaw > 90:
        push_factor.append(1 - (((1 / 90) * yaw) - 1))
    elif yaw < -90:
        push_factor.append(-1 - (((1 / 90) * yaw) + 1))
    else:
        push_factor.append((1 / 90) * yaw)
car.add_impulse(carla.Vector3D(x=impulse_strength * push_factor[0],
                                                    y=impulse_strength * push_factor[1],
                                                    z=0.0))
```

I can look at it a bit closer (if I have more time) if that doesn't fit the problem.

MarcelPiNacy, 2024-05-03:
Hi,

Sorry for the delay, is this issue still relevant?

Marcel