carla-simulator/carla #7210: Town03_Opt map crashes in map loading

anrp-tri, 2024-03-05:
CARLA version: 0.9.15
Platform/OS: Ubuntu 20.04
Problem you have experienced: TM crash on Town03_Opt
What you expected to happen: No TM crash
Steps to reproduce:
```
~/tmp/c0.9.15/n$ PYTHONPATH=PythonAPI/carla/dist/carla-0.9.15-py2.7-linux-x86_64.egg python2 PythonAPI/util/config.py -m Town03_Opt
load map 'Town03_Opt'.
~/tmp/c0.9.15/n$ PYTHONPATH=PythonAPI/carla/dist/carla-0.9.15-py2.7-linux-x86_64.egg python2 PythonAPI/examples/generate_traffic.py 
Segmentation fault
```
I was able to capture a stacktrace: `PYTHONPATH=PythonAPI/carla/dist/carla-0.9.15-py2.7-linux-x86_64.egg gdb --args python2 PythonAPI/examples/generate_traffic.py `

[stacktrace.txt](https://github.com/carla-simulator/carla/files/14500197/stacktrace.txt)

anrp-tri, 2024-03-12:
Per a discord conversation, erasing `Unreal/CarlaUE4/Content/Carla/Maps/TM/Town03_Opt.bin` and letting libcarla regenerate it seems to resolve this.

NnaYelsel, 2024-03-13:
Hello @anrp-tri , I got also a problem with Town03_Opt on windows, can I try also this solution ?
When you say letting libcarla regenerate it, you mean that I need to launch a "make LibCarla" after deleting the file. 
Can I directly make a make launch / make package after that ? 

Best regards

anrp-tri, 2024-03-13:
You can erase the file within your content folder, and when a TM requests it the libcarla client will re-create the binary map file.

NnaYelsel, 2024-03-13:
did worked, still have:

Assertion failed: px != 0, file C:/Users/Public/Public_CARLA/CARLA_source_code/carla/Build/boost-1.80.0-install/include\boost/smart_ptr/shared_ptr.hpp, line 783

When trying to launch a simulation. I will use Town03 instead of Town03_Opt

closed 2024-04-19 by Madecu