carla-simulator/carla #7211: fatal error LNK1120: 451 unresolved externals when make PythonAPI

kmsgm, 2024-03-06:
Before the recent Git repo update, the build was working fine. Now that the Git repo has been updated, I keep getting link errors when I try to build again. I've read that this can happen if you have multiple versions of python, but I only have one version of python.

CARLA version: 0.9.15
Platform/OS: Window 10, python 3.8, Visual Studio 2019
Problem you have experienced: Build error during `make PythonAPI` after git updated
What you expected to happen: Build `.whl` file in dist directory
Steps to reproduce: `make PythonAPI`

```
build\lib.win32-3.8\carla\libcarla.cp38-win_amd64.pyd : fatal error LNK1120: 451 unresolved externals
error: command 'C:\\Program Files (x86)\\Microsoft Visual Studio\\2019\\Community\\VC\\Tools\\MSVC\\14.29.30133\\bin\\HostX86\\x86\\link.exe' failed with exit status 1120
```

evoevo07, 2024-03-25:
I am working on the same issue...
Do you got a dist folder after run PythonAPI?
(in PythonAPI/carla)

xavisolesoft, 2024-04-19:
Did you check the comments described on this issue?
https://github.com/carla-simulator/carla/issues/3621
Maybe some of the described fixes will work for you.

ronitsharan, 2024-05-31:
I tried everything, but nothing worked for me