carla-simulator/carla #7213: Add info about supported python version

RobertoD91, 2024-03-06:
Windows release version 0.9.15 include .whl/.egg only for python 3.7.

On latest python (like 3.12) installing carla from pip will fail.

If i understood correctly only version 3.7 and 3.8 of python are supported.

Maybe we can add this information in the readme or in the PythonAPI directory?

According to https://devguide.python.org/versions/ version 3.7 is at its End Of Life (EOL), and 3.8 is in a 'security-only' phase, meaning no more binaries are released. This makes installation on Windows a bit more complex.

xavisolesoft, 2024-04-26:
Thank you for your feedback!
We created a ticked to build python API for more python versions. Will fix this soon.

https://github.com/carla-simulator/carla/issues/7473

closed 2024-04-26 by xavisolesoft

gitguige, 2024-09-12:
is this fixed