carla-simulator/carla #7223: How to save multiple sensor's data to disk simultaneously? (I need to save Raw, depth and semantic segmentation images of every frame in autopilot mode, to disk)

bsandy1978, 2024-03-07:
CARLA version: 0.9.15
Platform/OS: Windows

I need to save Raw, depth and semantic segmentation images of every frame, in autopilot mode, to disk. 
I need a python script that does the same. I can save Raw RGB images to disk, but I need the images of other sensors too, to save simultaneously in separate folders.

PatrickPromitzer, 2024-03-26:
For your case, reading the examples would help.
https://github.com/carla-simulator/carla/blob/master/PythonAPI/examples/sensor_synchronization.py

This example is using
``` python
blueprint_library = world.get_blueprint_library()
cam_bp = blueprint_library.find('sensor.camera.rgb')
lidar_bp = blueprint_library.find('sensor.lidar.ray_cast')
radar_bp = blueprint_library.find('sensor.other.radar')
```

How you want to save your sensor data after you get them, is your choice.

JoseM98, 2024-04-26:
Hi @bsandy1978, have you been able to solve your problem with what PatrickPromitzer has commented?

JoseM98, 2024-05-03:
As there has been no response I close the issue. Feel free to reopen it.