carla-simulator/carla #7224: Tearing image from camera sensor

silencial, 2024-03-07:
I am running Carla in synchronous mode with `fixed_delta_seconds=0.05`. I attached a camera sensor on the vehicle to collect RGB images. The image size is 640*480. The image is saved as a nump array and visualized as below:
```python
img = np.frombuffer(img.raw_data, dtype=np.dtype('uint8'))
img = img.reshape(640, 480, 4)
img = img[:, :, 2::-1]  # BGRA to RGB
plt.imshow(img)
```
 However some images have this tearing problem as shown below:
[image]
What could be wrong here and how should I fix this?

CARLA version: 0.9.15
Platform/OS: Ubuntu 20.04
Problem you have experienced: Tearing RGB image
What you expected to happen: RGB image rendered correctly
Steps to reproduce:
Other information (documentation you consulted, workarounds you tried):

PatrickPromitzer, 2024-03-26:
You can try saving the numpy array and see if it has the same problem.
```
# pillow
from PIL import Image as PILImage

cam_image = PILImage.fromarray(np_array)
cam_image.save(file_name) # change file_name
```

You can do the same with the image class from calra

``` python
img.save_to_disk(file_name)
```
https://carla.readthedocs.io/en/latest/python_api/#carlaimage

JoseM98, 2024-04-26:
Hi @silencial, have you been able to verify that the saved images or saved data have the same tearing problem?

silencial, 2024-04-27:
Actually I was saving the data with the `PIL.Image.fromarray()` method and plotted it afterwards. And the tearing problem was there.
I have not tried saving the image directly with the CARLA method. Will test later.

JoseM98, 2024-05-24:
I am closing this issue as there has been no more feedback. If you try the CARLA method and it still doesn't work, feel free to reopen the issue.

closed 2024-05-24 by Aperiss